360Controller/360Controller #117: Uninstall Xbox 360 Controler Drivers

asoksevil, 2015-07-11:
How can you uninstall it?

I have both the Xbox 360 and One controller and although the X360 is the wireless one with the USB cable, it doesn´t detect it so decided to go with the One that works flawlessly.

Thanks!

FranticRain, 2015-07-11:
Just a note, unlike the Xbox One controller, plugging the play and charge kit into an Xbox 360 controller does not turn it into a wired controller. It is still a wireless controller and requires a wireless dongle to work. The beta version of this driver includes both Xbox One controller support and a built in uninstaller. You can check that out here: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview6.pkg

asoksevil, 2015-07-12:
Thanks for the fast reply.

As I already have "Xone Controller", which one do you recommend me to use it?

Also, I am trying to find the uninstaller but can´t find it...

FranticRain, 2015-07-12:
Use this driver. It has more features than mine.

To uninstall "Xone-OSX" there is an uninstall.command included in the downloaded dmg. If you no longer have it, download the latest version and run the uninstaller from there. As for the 360Controller, simply install the new version over the old version. No uninstall required.

Once the new version is installed, the uninstall button for that driver is available in the bottom left hand corner of the advanced tab.

asoksevil, 2015-07-12:
Got it thanks!

Works great for OpenEmu.

jonathanhammond, 2015-07-17:
Hi i'm having issues with the builtin uninstaller on Install360Controller0.15Preview7 . Any advice?
[image]

RodrigoCard, 2015-07-22:
@jonathanhammond disconnect your controller from your computer, restart, and then try again, it should work!

asoksevil, 2015-07-22:
BTW, I have noticed that I am able to use the joy sticks on menus but I can´t use them outside of them while the D Pad works flawlessly on both cases.

I was able to replicate this issue using both Xone Controller and Xbox 360 Controler

RodrigoCard, 2015-07-28:
there are issues in the uninstaller for latest Yosemite (10.10.4) And probably El Captain.
Fixed it in 64be136

please check the latest version
 https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

closed 2015-08-19 by RodrigoCard