360Controller/360Controller #118: Fifa 13 not supporetd.

ichu501, 2015-07-13:
First of all thanks to give the wonderful support.I need to use my 360 controller for FIFA 13 app, If i try to connect app will crash or the device not detected,Please give me a solution.

MaddTheSane, 2015-07-13:
Sadly, there's not much we can do. If the game doesn't support game controllers at all, installing this driver won't make games automatically start working.

RodrigoCard, 2015-07-13:
Yes. As @MaddTheSane said, There is nothing much we can do. This is probably a problem on the game itself.

Anyway, is it a windows version with a wrapper or something like that?

RodrigoCard, 2015-07-28:
I will close this one, A quick search revealed this is a windows game unofficially ported.
So, a tip, in this screen:
[image]
Google for "windows xbox controller mapping ids" or something like that and swap the button ids for the same as windows' button ids, I guess it should mostly work :)

closed 2015-07-28 by RodrigoCard