360Controller/360Controller #119: Controller not working if connected after startup OS X 10.11 Public Beta

redeemer666, 2015-07-26:
Hello,

I'm having some issues with the driver on the OS X 10.11 Public Beta.
If I connect the Xbox 360 Controller (I have only wired one, so don't know if it's the same for the wireless) when the computer is booted, the preference pane recognize it but doesn't work. If I connect it before booting up the mac, everything is fine but the issue reappear if i disconnect and reconnect it.
Console keeps sending this kind of messages when issue appear:

26/07/15 09:55:11,640 System Preferences[488]: Failed to connect (bindingInputColumn) outlet from (Pref360ControlPref) to (NSTableColumn): missing setter or instance variable
26/07/15 09:55:11,640 System Preferences[488]: Failed to connect (linearDeadzoneLeft) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
26/07/15 09:55:11,640 System Preferences[488]: Failed to connect (linearDeadzoneRight) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
26/07/15 09:55:11,640 System Preferences[488]: Failed to connect (xoneRumbleOptions) outlet from (Pref360ControlPref) to (NSPopUpButton): missing setter or instance variable
26/07/15 09:55:11,642 System Preferences[488]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fc0d66125b0 H:[MyWhole360Controller:0x7fc0d6612680(465)]>",
    "<NSLayoutConstraint:0x7fc0d6616030 H:|-(20)-[MyWhole360Controller:0x7fc0d6612680]   (Names: '|':NSView:0x7fc0d3c94320 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc0d662d6e0 h=-&- v=-&- H:|-(0)-[NSView:0x7fc0d3c94320]   (Names: '|':NSTabView:0x7fc0d3cc72e0 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc0d662d730 h=-&- v=-&- H:[NSView:0x7fc0d3c94320]-(0)-|   (Names: '|':NSTabView:0x7fc0d3cc72e0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc5c0 H:[NSTabView:0x7fc0d3cc72e0]-(0)-|   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc080 H:|-(0)-[NSTabView:0x7fc0d3cc72e0]   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d6615e60 H:[MyWhole360Controller:0x7fc0d6612680]-(8)-[NSBox:0x7fc0d6421950'Left Stick']>",
    "<NSLayoutConstraint:0x7fc0d6428850 H:[NSBox:0x7fc0d6421950'Left Stick']-(20)-|   (Names: '|':NSView:0x7fc0d3c94320 )>",
    "<NSLayoutConstraint:0x7fc0d6427490 H:[NSBox:0x7fc0d6421950'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fc0d3ccbeb0 H:[NSBox:0x7fc0d3cc8810'Box']-(0)-|   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc3c0 H:|-(0)-[NSBox:0x7fc0d3cc8810'Box']   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccb150 H:[NSBox:0x7fc0d3cc8810'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fc0d66125b0 H:[MyWhole360Controller:0x7fc0d6612680(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
26/07/15 09:55:11,649 System Preferences[488]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fc0d66125b0 H:[MyWhole360Controller:0x7fc0d6612680(465)]>",
    "<NSLayoutConstraint:0x7fc0d6616030 H:|-(20)-[MyWhole360Controller:0x7fc0d6612680]   (Names: '|':NSView:0x7fc0d3c94320 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc0d662d6e0 h=-&- v=-&- H:|-(0)-[NSView:0x7fc0d3c94320]   (Names: '|':NSTabView:0x7fc0d3cc72e0 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc0d662d730 h=-&- v=-&- H:[NSView:0x7fc0d3c94320]-(0)-|   (Names: '|':NSTabView:0x7fc0d3cc72e0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc5c0 H:[NSTabView:0x7fc0d3cc72e0]-(0)-|   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc080 H:|-(0)-[NSTabView:0x7fc0d3cc72e0]   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d6615e60 H:[MyWhole360Controller:0x7fc0d6612680]-(8)-[NSBox:0x7fc0d6421950'Left Stick']>",
    "<NSLayoutConstraint:0x7fc0d6428850 H:[NSBox:0x7fc0d6421950'Left Stick']-(20)-|   (Names: '|':NSView:0x7fc0d3c94320 )>",
    "<NSLayoutConstraint:0x7fc0d6427490 H:[NSBox:0x7fc0d6421950'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fc0d3ccbeb0 H:[NSBox:0x7fc0d3cc8810'Box']-(0)-|   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc3c0 H:|-(0)-[NSBox:0x7fc0d3cc8810'Box']   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccb150 H:[NSBox:0x7fc0d3cc8810'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fc0d66125b0 H:[MyWhole360Controller:0x7fc0d6612680(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
26/07/15 09:55:11,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:11,930 System Preferences[488]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fc0d66125b0 H:[MyWhole360Controller:0x7fc0d6612680(465)]>",
    "<NSLayoutConstraint:0x7fc0d6616030 H:|-(20)-[MyWhole360Controller:0x7fc0d6612680]   (Names: '|':NSView:0x7fc0d3c94320 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc0d662d6e0 h=-&- v=-&- H:|-(0)-[NSView:0x7fc0d3c94320]   (Names: '|':NSTabView:0x7fc0d3cc72e0 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc0d662d730 h=-&- v=-&- H:[NSView:0x7fc0d3c94320]-(0)-|   (Names: '|':NSTabView:0x7fc0d3cc72e0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc5c0 H:[NSTabView:0x7fc0d3cc72e0]-(0)-|   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc080 H:|-(0)-[NSTabView:0x7fc0d3cc72e0]   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d6615e60 H:[MyWhole360Controller:0x7fc0d6612680]-(8)-[NSBox:0x7fc0d6421950'Left Stick']>",
    "<NSLayoutConstraint:0x7fc0d6428850 H:[NSBox:0x7fc0d6421950'Left Stick']-(20)-|   (Names: '|':NSView:0x7fc0d3c94320 )>",
    "<NSLayoutConstraint:0x7fc0d6427490 H:[NSBox:0x7fc0d6421950'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fc0d3ccbeb0 H:[NSBox:0x7fc0d3cc8810'Box']-(0)-|   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccc3c0 H:|-(0)-[NSBox:0x7fc0d3cc8810'Box']   (Names: '|':NSView:0x7fc0d3c94ba0 )>",
    "<NSLayoutConstraint:0x7fc0d3ccb150 H:[NSBox:0x7fc0d3cc8810'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fc0d66125b0 H:[MyWhole360Controller:0x7fc0d6612680(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
26/07/15 09:55:11,936 System Preferences[488]: No value for RumbleType
26/07/15 09:55:12,001 System Preferences[488]: Result of kextstat = 0
26/07/15 09:55:12,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:13,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:14,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:15,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:16,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:17,000 kernel[0]: send - failed to start (0xe000404f)
26/07/15 09:55:17,000 kernel[0]: send - failed to start (0xe000404f)
26/07/15 09:55:17,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:18,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:19,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:20,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:21,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:22,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:23,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:24,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:25,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:26,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:27,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:28,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:29,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:30,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:31,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:32,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:33,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:34,000 kernel[0]: Failed to send special message 1e
26/07/15 09:55:35,000 kernel[0]: Failed to send special message 1f
26/07/15 09:55:36,000 kernel[0]: Failed to send special message 1e

In loop.
This happens with 0.14 and 0.15beta7

RodrigoCard, 2015-07-28:
I changed some constraints last night. Does this fix the problem?
https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

redeemer666, 2015-07-28:
Hello Rodrigo,

thank you for your quick answer. The new version is a no go for me, same issue as before and now the LED on the controller doesn't turn on when recognized by the driver.

About the LED, I forgot to tell you that, when the problem happen, the LED doesn't turn on, and the preference pane, though it recognize the controller, doesn't respond to any button press or feedback on the sticks.

RodrigoCard, 2015-08-19:
See #127

RodrigoCard, 2015-09-01:
It is not a duplicate. Sorry.
Do you still have this problem in the latest beta (both the driver and the OSX)?

redeemer666, 2015-09-01:
Hello Rodrigo. With latest beta wired controller works perfectly good. Status LED is on and controller is recognized even when you connect with mac online. Console not reporting any error message. FYI, I'm on OS X 10.11 PB6, just updated. Thank you for solving this.

closed 2015-09-10 by RodrigoCard