360Controller/360Controller #121: What about UnTattie?

Mstrodl, 2015-07-27:
I need to uninstall this as it caused some... issues in El Capitan (OS X 10.11). Where is the UnTattie file?

FranticRain, 2015-07-27:
What are your issues with El Capitan? If you are vague about it, we can't fix the issues. As for an uninstaller, the preview builds of the next version of the driver include an uninstaller built into the preference pane. Simply install this version: https://dl.dropboxusercontent.com/u/14890209/Install360Controller0.15Preview8.pkg over your previous installation, and then open up the preference pane and use the uninstall function in there.

RodrigoCard, 2015-07-28:
there are issues in the uninstaller for latest Yosemite (10.10.4) And probably El Captain.
Fixed it in 64be136044c57d5cd3d84bfd2531c19ea076e687

RodrigoCard, 2015-07-28:
Does this one works correctly for you @Mstrodl ? https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

Mstrodl, 2015-07-28:
Yup, Don't know why but it did :+1:

Mstrodl, 2015-07-28:
Well, My issues were more that for some reason it didn't detect the controllers (I don't know if I used the wrong installer or what)

closed 2015-07-28 by Mstrodl