360Controller/360Controller #123: Controller Instantly Disconnects (10.11 Beta 5)

iCraftSn0w, 2015-07-28:
So I'm using the El Capitan beta, and I'm trying to use this driver with my Xbox 360 controller and when I plug in the controller the actual centre lights illuminate briefly but then turn off. I can tell the controller is working for a moment or two because if I move any of the analog sticks or press any of the buttons the UI on the screen reacts accordingly but then stays like that even if i move it. Any ideas?

RodrigoCard, 2015-07-28:
does this one helps? https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial

Apparently, #119 is the same problem

iCraftSn0w, 2015-07-28:
Yeah this is the first driver I have tried and then tried the previous version, however both of them have the same issue. Perhaps it's just an issue with El Capitan?

RodrigoCard, 2015-07-28:
Probably!

I will tell you if I got time to test El Capitain

BSVogler, 2015-07-31:
I reported same issue here:
https://github.com/electric-monk/360Controller/issues/8

RodrigoCard, 2015-08-19:
duplicate of #127

RodrigoCard, 2015-09-01:
It is not a duplicate. Sorry.
Do you still have this problem in the latest beta (both the driver and the OSX)?

DrDomVonDoom, 2015-09-10:
I have this issue as well. Not sure why, it was working fine on the last patch, I didn't use it again for some time, then overnight something happned and it suddenly stopped working. Im not on El Capitan either, Im running Yosemite 10.10.2

iCraftSn0w, 2015-09-10:
Ok so a little update. I've just got a new MacBook Pro to use and I have installed this driver and then I installed the El Capitan beta and it seems to be working? No disconnecting issues or anything, I was playing Deus Ex for a solid two hours. So I'm not sure how this can benefit you but I thought it might be relevant.

DrDomVonDoom, 2015-09-10:
Thanks, Im not using the beta, Im actually still on Yosemite, tried uninstalling and reinstalling but nothing. Since I am using Steam in home streaming from my gaming PC, I think for now I’ll end up using a wireless PC adaptor and use it that way

Thanks tho,

ellioman, 2015-10-01:
Just to share my experience. Just updated to El Capitan and the driver no longer worked, couldn't find the controller. I then installed the 0.15 (beta 3) and everything is now working fine after restarting the computer.

RodrigoCard, 2016-01-19:
i think it is ok to close this

closed 2016-01-19 by RodrigoCard