360Controller/360Controller #126: Hori Xbox one Real Arcade Pro support

tcook1231, 2015-07-31:
Hey guys. I just installed this driver and I was trying to get my stick working with my mac. 
Its IDs look like the following
Product ID:	0x0063
Vendor ID:	0x0f0d  (HORI CO., LTD.)

Any help would be greatly appreciated.

FranticRain, 2015-07-31:
For some reason this device is in the Info.plist with the wrong Product and Vendor ID's. Probably my doing. "HoriRAPHayabuseXboxOne" idProduct probably shouldn't be 62722. I have no idea what caused that, but blame says I did it.

@RodrigoCard If you could put the correct Product ID in there for me, that would be great.

tcook1231, 2015-07-31:
Thanks man.

tcook1231, 2015-08-02:
Could I edit the source myself and see if it works? What do I need to change?

FranticRain, 2015-08-02:
The driver would need to be recompiled for it to take effect, so no you can not.

tcook1231, 2015-08-02:
Fair enough.

RodrigoCard, 2015-08-02:
I will take a look :)

tcook1231, 2015-08-02:
Man you just made my day. This driver kicks the pants off of any other controller driver I have ever used. You guys are really doing something right with this. Thank you for all the hard work that has gone into this project

tcook1231, 2015-08-02:
just out of curiosity what would goes into adding a device? is it just putting in the product and vendor ids?

FranticRain, 2015-08-02:
Yeah. As long as its an Xbox One, Xbox 360, or Xbox Original controller, its just that. Super easy. Unfortunately for you however, RodrigoCard is a busy man.

tcook1231, 2015-08-02:
Fair enough. I don't mean to come off as inpatient. I am sure it will be worth the wait.

tcook1231, 2015-08-02:
So what should the values look like?

FranticRain, 2015-08-02:
You just convert the values from hexidecimal to decimal.

Product ID: 0x0063 = 99
Vendor ID: 0x0f0d = 3853

tcook1231, 2015-08-02:
got it. I'm just a mechanical engineer trying to do a bit of code. I haven't had to break out the hammer yet so there is still hope

FranticRain, 2015-08-02:
If you're looking to hack this out and compile it yourself, you'll want to stop right there. In order to compile kexts for use on your machine, you have to have a very special signing certificate from Apple. I know both RodrigoCard and I have them, but you have to go through this crappy process to get one. There is a way to disable that signing restriction, but that involves changing your boot-args, which I cannot recommend you do.

tcook1231, 2015-08-02:
Fair enough. So I can't just gut and change things. I'll stop there.

FranticRain, 2015-08-02:
Heres a version I signed. Its got some weird beta features in it, and doesn't support force feedback, but put it into "/Library/Extensions" and the reboot and it might work? I'm not sure. https://dl.dropboxusercontent.com/u/33409904/360Controller.kext.zip

tcook1231, 2015-08-02:
I'm not gonna bother. It's not worth the headache. I don't want to trust might work...

tcook1231, 2015-08-04:
@FranticRain @RodrigoCard  I was wondering if either of you guys could get the southpaw support into a dmg or pkg.

RodrigoCard, 2015-08-28:
I will make a build with these and other changes. :)
Also, I see that this device's IDs is in the list. So, I will close this.

closed 2015-08-28 by RodrigoCard

RodrigoCard, 2015-08-28:
Please install the latest version, https://github.com/d235j/360Controller/releases/tag/v0.15_beta2_unofficial Cheers!