360Controller/360Controller #127: El Capitan support

MaddTheSane, 2015-08-02:
It seems that Apple changed the driver implementation for USB kernel drivers in El Capitan, as Xcode 7 beta 4 doesn't build the kexts when building against the 10.11 SDK.

Other than a lot of warnings about virtual declarations in subclasses, the headers `IOKit/usb/IOUSBDevice.h` and `IOKit/usb/IOUSBInterface.h` aren't present. The headers may have been renamed, but it's possible that there is  replacement that Apple would want us to use.

It seems like there are `IOUSBHostInterface.h` `IOUSBHostDevice.h` and `IOUSBHostHIDDevice.h`. Although I don't know how they relate to the older headers.

FranticRain, 2015-08-03:
They probably want us to use "IOUSBHIDDriver," but legacy game support is abysmal.

MaddTheSane, 2015-08-03:
Looking at the code, it does seem like IOUSBHostDevice and IOUSBHostInterface replace IOUSBDevice and IOUSBInterface, respectively. And a lot of the methods are re-done, with the first character being lower-case instead of upper-case. Still, updating it for El Capitan will be a bit more involved than that.

And it looks like even IOUSBHIDDriver was renamed as IOUSBHostHIDDriver.

FranticRain, 2015-08-03:
@MaddTheSane Oh wow, jeez. They really overhauled the whole system. Crazy.

MaddTheSane, 2015-08-03:
The good news is that all the deprecated functions listed in the new headers (commented out) also list their replacements.

RodrigoCard, 2015-08-03:
Apple really hates Backwards compatibility, doesnt it? -.-

MaddTheSane, 2015-08-17:
My work so far can bee seen here: https://github.com/MaddTheSane/360Controller/tree/elCap
I haven't found a good replacement for `IOUSBFindEndpointRequest` yet :worried:

RodrigoCard, 2015-08-19:
Thank you for your work @MaddTheSane !!
What does IOUSBFindEndpointRequest do?

MaddTheSane, 2015-08-19:
`IOUSBFindEndpointRequest` was used to eventually get the proper Pipe object. I'm fairly certain that `getInterfaceDescriptor` and `getNextAssociatedDescriptorWithType` will be needed based on header documentation for `IOUSBHostInterface::FindNextPipe(IOUSBHostPipe* current, IOUSBFindEndpointRequest* request)`. I did push a possible fix for the wired controller, but I have no El Cap devices to test it on.

Roy-Hub, 2015-08-19:
Guys I am updating a driver for El Capitan in xcode 7
1) I updated `IOUSBHIDDriver.h` to `IOUSBHostHIDDevice.h` 
2) My code was using `IOUSBInterface`, which was previously defined in `IOUSBDevice.h`. Now the definition was moved to `IOUSBUserClient.h`
So, i included `IOUSBUserClient.h` in my code

[Reference - Mac Ref Lib - Pre Release ](https://developer.apple.com/library/prerelease/mac/releasenotes/General/APIDiffsMacOSX10_11/Objective-C/Kernel.html)

**The Problem - redefinitions** 
For example 

`kUSBHIDDesc` (and many others) are defined in `IOUSBHostHIDDevice.h` 
And `IOUSBUserClient.h` internally includes -> `IOKit/usb/USB.h`
The following lines of code is from `IOKit/usb/USB.h`
```
#if !defined(__USB__)
#    include <IOKit/usb/USBSpec.h>
#endif
```
And `kUSBHIDDesc` is re-defined in `USBSpec`.

So I believe its not possible to include `IOUSBHostHIDDevice.h`  and `IOUSBUserClient.h` simultaneously. Do you guys, have any idea on how to proceed? 
 I am a newbie in driver programming. I wish someone could give me some help. Thanks.

RodrigoCard, 2015-08-24:
Important question, does this new IOUSBHostInterface works in an older version of OS X?
In which version was that introduced?

RodrigoCard, 2015-08-27:
Reading this https://developer.apple.com/library/prerelease/mac/releasenotes/Darwin/RN_USB/Articles/10_11Changes.html#//apple_ref/doc/uid/TP40007739-CH3-SW1

"Applications and third-party drivers for vendor-specific USB devices built with 10.10 SDK require no modification." - 
So, the currently builds should work? If not, why?
Is it possible to build it using the 10.10 SDK in xcode 7?

I am asking this because looks like the new API will, (probably?) not work in older versions. And I think it is too early to drop support for older OS X > 10.9

--------
That said, any ideas? Maybe a compatibility layer/wrapper classes?

RodrigoCard, 2015-08-31:
@MaddTheSane Could you also make a pull request with your work? Thanks! :+1:

RodrigoCard, 2015-09-01:
Nevermind all those questions, I have been testing stuff in El Capitain today and the driver does work.
I though it doesn't ... So people having problem with El Captain is unrelated...
By the way, this work will probably be needed in the future.
Please tell me if you need help testing.
Cheers!

MaddTheSane, 2015-09-01:
On a side note, I do see rootless mode not liking the installer script creating a system link in the */System/Library/Extensions/* folder. Does anyone know if the installer complains about it, either failing the install or complaining in the system console? Would the symlink in */System/Library/Extensions/* be needed in El Cap (or even recent Yosemite builds)?

RodrigoCard, 2015-09-02:
Well, I couldn't test the Force feedback in El Capitain, but if it works it shouldn't be needed anymore.
Do you know why the it doesnt work on the preference pane anymore?

klopfdreh, 2015-09-19:
Ok, so If I understood the content of this issue right - the driver will run with 10.11 - thats really good, because I am happy to get split screen support and just want to install el captain just in time when it is released! :smile:

tmecklem, 2015-09-22:
I have two xbox (original) controllers with a USB adapter, and both worked fine before an update to El Capitan. After the update, however, one controller stopped being recognized. The console output is: 

kernel[0]: Unknown device release 121
kernel[0]: start - unable to find the interface
System Preferences[671]: No devices found? :( device count==0, i==0

I thought this was strange, so I tried swapping the adapters. It didn't matter which adapter or USB ports I chose, the same controller kept giving that message while the other worked fine.

I finally tried it on a different mac I have with the same results. Pre-upgrade, both controllers worked. After the El Capitan, the same controller refused to work. What's up with that? How can I help troubleshoot this?

RodrigoCard, 2015-09-22:
Hello @tmecklem Could you create a new issue for that, please?
I think @kasbert may be able to help you, as he is the guy who added Xbox original support for this driver :)

klopfdreh, 2015-09-30:
Hi, I just installed OS X 10.11 and I can confirm the driver is working for usb xbox 360 controller. All is working like expected - even the rumble mechanism

lostincreation, 2015-10-16:
Hi, i have the latest Xbox controller (wired) for Windows and El Capitain. When i install the Beta3, there is no device recognition. I used to have it worked, which release should do it?

kasbert, 2015-10-31:
Sadly, my chinese clone Xbox Original Controller stopped working too with El Capitan upgrade. Could be the illegal device:vendor ids (ffff:ffff). Because of the kext signing requirement, I am not very interested in debugging the problem. Maybe I just buy a 360 controller.

RodrigoCard, 2015-11-02:
@kasbert Does the regular (original) xbox original controller still works?

RamblinLamb, 2015-11-02:
My OEM Xbox 360 game pad does not work on Mac OS X El Capitan with version
0.15.

-- 
Richard T. Lamb
*Ramblin Lamb Photography*
35905 SE Boulder Ct
Snoqualmie WA 98065
rtlamb@ramblinlamb.com
Http://www.ramblinlamb.com <http://www.ramblinlamb.com/>
B - 425.427.2573
C - 425.444.6243
F - 425.831.6840

kasbert, 2015-11-03:
@RodrigoCard I don't have an original (Microsoft) xbox controller. My second controller is a xbox 360 controller and that works ok.

tburazin, 2016-02-19:
@RodrigoCard I know this issue is closed but just for posterity, I'm running El Capitan 10.11.3 and my original Xbox controller S works. I can get my hands on a "Duke" to test if you'd like.

CorbMax, 2016-08-18:
Hi... I'm trying to pair my Original Xbox 360 wireless controller with my MBP 2011 with El Capitan 10.11.6 using an [unofficial wireless receiver] (https://www.amazon.it/gp/product/B00MUUZDMS/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1), but... no luck.
Driver installation is good. I can find icon in System Preferences, but trying to pair device green led on wireless receiver start blinking forever, just like green led on wireless controller... spinning forever.
Any news on this issue?

FranticRain, 2016-08-18:
@CorbMax Your issue is totally unrelated to this issue. Please create a new issue so we can help you solve your problem.

kasbert, 2016-08-19:
@CorbMax I bought one of those unofficial receivers, too. It was horrible. There's no way to fix it in the driver and I bought an official receiver.

FranticRain, 2016-08-21:
I'd like to actually revive the real issue here. I emailed some guys from Apple regarding issues I have been having with the consistency of error reporting within the kernel, and they seem to think it is because we are using the old shim IOKit classes. @MaddTheSane's code is really close, but its got some issues I can't figure out on my own. Currently, the elCap branch will compile, but it won't load. This is fixed by a simple change to the OSBundleLibraries in Info.plist. You can find what needs to go in there by using 'kextlibs' on 360Controller.kext. I'll put in a pull request to fix that in a minute. The main problem arises with the fact that the `provider` object in `start` isn't actually of type `IOUSBHostDevice`, which means that the OSDynamicCast immediately fails. I have absolutely no idea how to find out what `provider` actually is, or if it is just a raw `IOService`. Do you guys have any ideas?

MaddTheSane, 2016-08-21:
It seems like the `IOProviderClass` key in the Info.plists are still using the old `IOUSBDevice` class instead of `IOUSBHostDevice`.

FranticRain, 2016-08-21:
Yeah, I just figured that out too. I made some changes to the way you found interfaces and pipes and got an Xbox One controller to work with the driver, so I'm going to try and clean up the rest of it and put in another pull request.

shinji257, 2016-09-13:
I thought to drop a line and mention that 0.16.4 seems to work fine with 10.12 Beta 7 Sierra.

wesen, 2017-07-20:
I'm interested in working a little bit on this driver to fix #357 which is driving me nuts. I was wondering which branch I should be working against, is @MaddTheSane 's elCap branch a good starting point still?

MaddTheSane, 2018-03-09:
Good news on my elCap branch: The 360 controller's LEDs blink when plugged in.
That's about it, though. SDL2 isn't finding it, which probably mens something got messed up when I migrated the code.