360Controller/360Controller #129: Unity Game Engine Button Mappings

gtjeffw, 2015-08-07:
Hello. I am wondering if it is feasible to create a feature that allows the 360Controller drivers to emulate the button mappings as if running on a Windows machine. Under Windows, the buttons map differently in Unity3D games than in OSX. Something like a "Unity/Windows compatibility mode" checkbox in the System Pref panel is what I am thinking would be useful. This feature would mainly help with Unity games where the developers have only partially supported OSX.

The differences are detailed here: http://wiki.unity3d.com/index.php?title=Xbox360Controller

I don't know what the underlying cause of this difference is, and it is likely impossible to address from the driver side, but I thought I'd ask. BTW, thanks for all the work on the driver!

Rcannon95, 2015-08-11:
I would also be interested in knowing if this is possible. SkyRogue and Megabyte Punch would be 2 example games.

tomwor, 2015-08-11:
I also just stumbled over this issue, would be great to have a solution.

erebuswolf, 2015-08-12:
I would love this feature as well! Are you guys able to get the inputs into Unity at all? I tried changing the bindings but I can't seem to get any input to be recognized by Unity off the latest release today :\

erebuswolf, 2015-08-12:
Never mind. Restarting and then having the controller plugged in before launching unity fixed it. The joystick has some weirdness to it compared to the windows driver though I may post about it when I have run some more tests.

RodrigoCard, 2015-08-19:
you may be able to partially achieve that by changing the buttons order in the binding tab. did you test that?

slawo, 2015-10-08:
I use unity and to address the differences between controller layouts on Windows, Mac Android, Linux and OS X there is the [Galant Games InControll plugin](http://www.gallantgames.com/pages/incontrol-introduction)

erebuswolf, 2015-10-08:
I did end up buying that plugin for windows stuff. But I think it would be
nice to fix here too. Thanks for posting a solution though!

slawo, 2015-10-08:
Changing the layout of the controllers now would break all existing apps which already have adapted to the existing mappings and has the potential to cause way more damage than it's worth.

As a reference, see the problem with the 1 axis for triggers in Windows, once this behaviour was well established microsoft had to wait the release of a new API to address this issue.

erebuswolf, 2015-10-08:
My suggestion was to have a checkbox in the driver to switch to a unity
compatible mode. An opt in option not a only option.

FranticRain, 2016-02-12:
Considering the recent discussion about how Unity is the one messing up the button IDs, can we even do anything? This one seems DOA because Unity is the problem here, not the driver. See #210

slawo, 2016-02-12:
A driver is **not** supposed to handle the quirks of each game engine... If you need to fix an issue which is due to a game engine it is completely insane to tell users to go in their driver preferences and check a box when they want to play your game... And uncheck it when they want to play another one.

If you attend interested in supporting OSX users then take the time to give them proper support instead of breaking a driver. For Unity Incontroll does the trick with near 0 constraints. If you want something more solid use Rewired...

erebuswolf, 2016-02-12:
Thanks for the suggestion with Rewired. It isn't uncommon for a driver to
handle to the quirks of game engines. Graphics drivers do this all the
time, it is literally what the nvidia geforce experience is a UI for. The
teams at Nvidia do game detection and custom patch in shaders at the driver
level to get games running better. But I understand your position in
wanting to simply do things a single right way and have everyone else
conform to that driver.

slawo, 2016-02-12:
I don't want to look like I'm raving, but I'd like to make a strong case concerning drivers and their role. 

Graphic drivers indeed implement a lot of fixes... But they respond to very different rules:

- they are a selling point for the manufacturer's hardware. More processing power sells better.
- they implement ever changing standards with a lot of undefined behaviour and with additional custom manufacturer extensions which also have often undefined edge cases.
- they are updated very frequently
- they must compensate for the shortcuts taken by the manufacturer and must keep compatibility with previous assumptions made by the game makers as to the results from some of their implementation quirks and extensions which have changed in the meantime.

Graphics drivers are actually a very good example of what a driver should not be. In fact, do you think we can call those things drivers anymore? The drivers are in there but when you install the packages you have 100s of MBs of software, profiles and other bloatware the average user should never even have to use or to see. When you have to configure your drivers before launching any kind of software you are basically back to the DOS era.. I didn't enjoy having to configure my drivers at the time, and I still don't like having to do that today.

Would you find it OK having to configure your mouse, sound card, keyboard, and storage drivers before getting a game to launch?

Also once a behaviour is implemented the developers rely on that behaviour to be consistent. And that's the reason there are so many profiles in the drivers. Some games expect a behaviour that has changed in the meantime to increase performance, so they need a given profile to avoid glitches or lost performances.

Would you write to sony to make them change the binding for the DualShock 4 because they don't match those of the DualShock 3? Would you require all european manufacturers to change the AZERTY keyboards to QWERTY because you expect users to have that configuration?

Sure our life would be so much easier if we had one standard and one behaviour for each device category, but it's not the case. And a driver is a driver you expect it to be consistent and to never fail at doing what it does. And when it's updated you should expect it to always give you the same results it used to, unless a documented behaviour did not work as expected.

Or maybe I'm getting old and I don't like change....

erebuswolf, 2016-02-13:
I see what you are saying, but at some point the rubber need to meet the
road. I want the xbox driver that will work with the most games out of box
or at least have settings that will allow me to make the driver work with
the game. If you rely on the game developers to correctly support a niche
controller driver for mac you will inevitably be unable to support games
that are no longer being patched and will likely not be compatible with a
lot of games moving forward that simply don't know this driver exists. The
use case for this driver is a gamer who is looking for a way to get a 360
controller working with mac games, putting the onus on the user to find out
how to get the controller to work with the game, therefor it makes sense to
give the user full control over how the driver works so they can configure
a version that works with the game they want to play. Very few developers
are going to assume mac users will have this driver or seek it out (though
obviously they should implement correct gamepad support for ps4 and other
controllers that have out of box drivers which this driver should be
compatible with).

I understand that I'm not contributing to this currently and so my opinion
should bear little weight, but that the usage page contains a comment of
"In some cases this may need an update from the manufacturer of the game or
a patched version." is a little crazy to me. I feel like the user should
not have to wait to be able to tweak button configs at the driver level to
get a game to recognize it correctly.

That's just my opinion and I do respect yours in what a driver should be.
But this isn't a driver that is shipping on macs or a driver that had first
party support from microsoft for the hardware. This isn't the driver that
all mac game devs will look at and say we need to make sure we work with
this because it's the standard. I wish that all games correctly handled mac
gamepad button configs but it seems silly to deny the users the ability to
fix it themselves when it is technically possible for them to do so.

slime73, 2016-02-13:
> If you rely on the game developers to correctly support a niche
controller driver for mac you will inevitably be unable to support games
that are no longer being patched and will likely not be compatible with a
lot of games moving forward that simply don't know this driver exists.

This is effectively the only viable xbox 360 driver for OS X. It's not niche, it's the only option. Game and engine developers either roll their own drivers or rely on users installing this driver, if they want to support Xbox 360 controllers in their OS X games.

Game or engine developers *must* implement OS X-specific joystick/gamepad code in order for *any* gamepad to work with their products in OS X. If they do that, Xbox 360 gamepads will work as long as users use this driver. It's worth noting that the SDL library (a library specifically for handling platform-specific differences in input and windowing so game developers don't have to) already has perfectly good support for 360 controllers when this driver is used. Steam and all Valve games and many non-Unity indie games use SDL in OS X.

This driver has been around for nearly a decade and its mappings have not changed in that duration. 
Unity has poor gamepad support / APIs basically across the board on all operating systems - the onus is on Unity to fix its own problems.

Adding a "Unity compatibility" button will cause many problems as soon as Unity improves its own situation, because then people using that option will complain that their mappings are wrong for new games which actually do the correct thing.

FranticRain, 2016-02-13:
Once again, we couldn't implement a "Unity Binding" button even if we wanted to because Unity is mangling it in the first place. Xbox One and 360 controllers appear to programs as identical controllers, but Xbox One controllers in Unity aren't messed up like the 360 ones are. There is no magic binding that fixes this issue, from what I've seen.

erebuswolf, 2016-02-13:
Fair enough. I'll keep the issue open as Unity needs to fix their shit.
Sorry I see I didn't structure my sentence well with my niche comment. I
meant using a xbox 360 controller on mac is a niche use case. Not that
there are many other driver options on mac and your driver is niche in that
category.

Mirageofmage, 2017-07-05:
There's actually a really nice [unity extension](https://github.com/JISyed/Unity-XboxCtrlrInput) you can use in place of unity's mangles system

FranticRain, 2018-06-01:
A note has been added to the Readme about this issue. As it is an issue with Unity, feel free to protest to them, but this isn't an actual issue with the driver, so I'm closing it.

closed 2018-06-01 by FranticRain