360Controller/360Controller #131: Turn off leds

floscr, 2015-08-08:
It would be nice if there was an option to turn of the led,
sometimes I play in a dark room and the LED is blinding.

appreciate all the work, thanks!

fent, 2015-09-08:
+1

I tend to disconnect my controller when not in use because of this

zero334, 2016-02-24:
O.k. I did some research how to achieve this.
According to: http://support.xbox.com/xbox-one/accessories/use-accessories-app-configure-elite-controller only the Xbox Elite Controller is capable of changing the LED (at least with the official app).
I don't know if this limitation is only set up by the windows app or is a real hardware thing.

Sadly I didn't find some information what we need to send to control the LED (brightness).
One way to figure out would be to capture the data send by the app to an Xbox Elite Controller.
I don't have an Elite Controller so I can't do it. Maybe someone with such a controller could do the capture.