360Controller/360Controller #133: Relic Hunters Zero support

s0nera, 2015-08-20:
It's not working with Relic Hunters Zero, but works with other games.

s0nera, 2015-08-20:
It is an open source game, and the code is here on github.
Maybe I the issue is with the game. Should I open it there?

FranticRain, 2015-08-20:
Uninstall my driver and install [this one instead](https://github.com/d235j/360Controller/releases/tag/v0.15_beta_unofficial). I have been moving my code into it because more games support this other driver. If that doesn't fix it, then it is definitely on the games end, and you should start an issue up with them.

FranticRain, 2015-08-20:
Yeah, if it doesn't work with this driver, then it most likely doesn't have controller support. You'd definitely want to check with them on this one.

s0nera, 2015-08-20:
Ok, I'll try. Thanks!

MATEUS M. S. BERGAMIN
19 9758 0051

lprunk, 2015-08-20:
Relic Hunters doesn't support gamepad on the mac, it's some issue with Game Maker. They used a custom xinput dll for windows and when on a Mac it changes to a default Game Maker library which is not guaranteed to work and since they don't have a mac to test it, they couldn't add full support to it. Relic Hunters is [open-source](https://github.com/mventurelli/relichunterszero), so he is encouraging people to mess with the game's code if someone has a solution for it =)

Source: One of the developers is sitting right next to me.

s0nera, 2015-08-20:
Well, I'll open this issue there, so maybe more people get interested.
And I'm mad I don't have coding skills to help solving this.

RodrigoCard, 2015-09-01:
Well, looks "solved" :)

closed 2015-09-01 by RodrigoCard