360Controller/360Controller #134: Digital currency donation address

felixvd, 2015-08-20:
I am so glad this driver has been updated so I can save on the 2 minutes to crash and restart every time I plug in my gamepad. 

I want to send someone who took part in this a coffee or something, preferably in digital currency so PayPal doesn't take half the money. Can I?

OKNoah, 2015-08-29:
PayPal doesn't take half.

felixvd, 2015-08-30:
PayPal would be a start, but 10% on a $3 coffee and >30% on anything under a dollar is a pretty big cut.

OKNoah, 2015-09-08:
I imagine exchanging $3 worth of Bitcoin for dollars is if not equally expensive then an extreme hassle or impossible. The go-to now seems to be Local Bitcoins which doesn't seem safe to me as people either scam on there or bring a knife or gun to the meeting and rob you of your Bitcoins and valuables. Exchanges will probably have a minimum withdrawal amount greater than $3, and there will be basic miner fees, BTC -> USD exchange fees and then withdrawal fees. I'd be interested to see how much of $3 you can get out and in what time span, if at all.

felixvd, 2015-09-09:
I didn't mean to make this question about the details of exchange services. All I'm saying is that I don't see a way to support this project currently, and PayPal fees [are definitely higher for small donations](https://www.reddit.com/r/dogecoin/comments/20pgnn/statistics_some_data_about_our_donation_drives/). There's no reason not to allow both options.

d235j, 2015-09-09:
Whom would such contributions go to? We have multiple developers, who are more or less involved in the project, and once we collect money, it will be tricky to decide whom it should go to and for what. I'm going to close this for now — if any of the other maintainers feel otherwise, please feel free to reopen, or email me directly.

closed 2015-09-09 by d235j

felixvd, 2015-09-09:
I'll just add that that is always a tricky question and usually best decided by the team. It could go to someone who has expenses (e.g. is donating web space) or who the team feels has contributed considerably, for example. I can see why not opening that can of worms would be easier, though.

A last note: all digital currency transactions are public and traceable if that's a concern.

RodrigoCard, 2015-09-10:
I agree with d235j, we are helping maintaining this but we dont do this for money (I believe other contributors agree), and mostly on our spare time. I dont think getting money from this is a good idea.