360Controller/360Controller #137: Xbox 360 wireless headset

Sk1nk, 2015-08-26:
I love the project, many thanks. Is there a possibility to add the Xbox 360 wireless headset as well (the one without bluetooth, witch needs to connect through the adapter)?

MaddTheSane, 2015-08-26:
Does it have a headphone-like jack?

As it is, even the wired headset isn't supported: the driver doesn't know how to get audio from the controller yet.

Sk1nk, 2015-08-27:
No it doesn't. Official support website ist this one: http://support.xbox.com/xbox-360/accessories/set-up-wireless-headset, it works in Windows so it might be possible.

RodrigoCard, 2015-08-31:
does it connect to the "Wireless gaming receiver" on windows or it connects in another way?

Sk1nk, 2015-08-31:
It does connect through the "wireless gaming receiver", even under OS X if I let both of them seek each other, they almost immediately stop looking for each other, but the headset doesn't connect.

MaddTheSane, 2015-09-01:
I can see it relying on the `WirelessGamingReciever` kext to connect, but needing a different driver class than `WirelessHIDDDriver`. It'd probably rely on `WirelessDevice` to set up the proper hooks.

Sk1nk, 2015-09-01:
I have no insight in programming, but if you need further tests, protocols or other insight, I am willing to help if you tell me how.