360Controller/360Controller #138: 0.15 Beta 2: Wrong 360 Wireless Button Mapping

felixbuenemann, 2015-08-29:
0.15 Beta 1 works fine with an original Wireless 360 Controller w/USB Dongle. But on 0.15 Beta 2 the DPad and most Buttons are recognized as DPad up, while the triggers and sticks are fine.

FranticRain, 2015-08-29:
This could be something with the remap functionality. It was just added to wireless 360 controllers in beta 2.

RodrigoCard, 2015-08-31:
I can confirm that. And when I try to remap, I get this:
[image]

Reseting the remap fixes that.

RodrigoCard, 2015-08-31:
@FranticRain could you look into that?

FranticRain, 2015-08-31:
I'll take a look at it, but I don't have any Xbox controllers on me right now, so it'll be a few days.

Duros, 2015-09-06:
I found that some game that native support the 360 controller have the start button the and right stick click mappen wrong, like Portal 2 and Lethal League. You have to press the right stick to get the start menu. I've got this with Beta 1 & 2.

FranticRain, 2015-09-07:
The original issue here is another settings bug. The default binding setup isn't loaded in through settings, causing all buttons to be bound to the 0th button, aka the d-pad up. I can do a couple things here. 1. Nothing. We really need to fix that settings bug. 2. Load the default mapping when the settings fail to load. This one is questionable because the only place to check it will be in the button press loop. And we would be adding a bunch of new cycles to every button press event. Something I would rather avoid. 3. I can disable rebinding of the Wireless 360 Controller until we fix the settings bug.

RodrigoCard, 2015-09-07:
3 is reasonable while we can't fix the settings bug, if possible.

Any ideas for the settings bug?

By the way, do you know why the wireless controller behaves differently
than wired one regarding this mapping issue?

FranticRain, 2015-09-07:
I've tried looking at that settings stuff before, and even looked for different ways to implement it. The real problem is that all of that stuff is so cryptic and horribly documented that I can't get any sort of grasp on it. As for the differences, I'm not sure why, but the properties of the wired controllers have their setProperties wireless controller's setProperties isn't. I don't know why it happens.

Xelnath, 2015-09-12:
I'm having this issue too. Pretty punishing - is there a workaround?

RodrigoCard, 2015-09-15:
ok, worked around https://github.com/d235j/360Controller/releases/tag/v0.15_beta3_unofficial

closed 2015-09-15 by RodrigoCard