360Controller/360Controller #139: device request: Hori Fighting Stick VX

city41, 2015-08-30:
Could you guys add the Hori Fighting Stick VX?

I notice "HoriFSVX" is in the plist file, but my stick has a different product and vendor id, and is not recognized by 0.15beta2.

Here's the stick info I get from my Mac:

**FIGHTING STICK VX:**
Product ID: 0x5502
Vendor ID: 0x24c6
Version: 5.33
Serial Number: 10300F9E
Speed: Up to 12 Mb/sec
Manufacturer: HORI CO.,LTD
Location ID: 0x14200000 / 1
Current Available (mA): 500
Current Required (mA): Unknown (Device has not been configured)

everblaze, 2016-01-19:
would it be possible for this device aswell?

Razer Atrox Arcade Stick:

  Product ID:	0x0a00
  Vendor ID:	0x1532
  Version:	1.00
  Serial Number:	00008108C0F50115
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Razer
  Location ID:	0x14200000 / 3
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

FranticRain, 2016-01-19:
I just added both of these devices. Let me know if they don't work. https://drive.google.com/file/d/0B5PLCKoMd4zGZmUxWWhiMGtFaDg/view?usp=sharing

closed 2016-01-19 by FranticRain