360Controller/360Controller #140: combine trigger axes into one axis

FactSphere, 2015-08-30:
Hi,

currently, the driver treats both triggers as separate axes. However, in Windows there is also a combined axis which is centered when no trigger is pressed. Assuming that both trigger axes go from 0 to 1, this axis gives `0.5*(1+right trigger - left trigger)`. This is very convenient to use for example as the yaw axis of an airplane.
Could this be incorporated into this driver as well? My proposal would be to give a boolean option in the Preference Pane, "link trigger axes", which sets both triggers to the output of the formula above using the `fiddleReport` function in `_60Controller.cpp`. 
In the game, one would then simply set the input to either one of the axes.
Unfortunately, I don't have an Apple developer account and couldn't figure out how to compile and install this thing from source without it, otherwise I would try to implement this myself...

RodrigoCard, 2015-08-31:
Interesting request, I always thought there was no use for that weird behavior on Windows Directinput implementation for xbox controllers. For me that was just something to force developers to use Xinput to get separate triggers. :P

slawo, 2015-10-08:
It's a horrible "feature".

There is no need to add another axis from the combined existing axes just for the Xbox controllers. This issue was fixed in the Xinput driver.

If both triggers work just like triggers on (let's say) PS3/PS4 controller then why would you want to break the consistency by adding something that is clearly trivial to do in software and at best just useless in most cases.

electric-monk, 2015-10-08:
I've had this request a number of times over the years, and I've always resisted it, for the reasons slawo states. As per the problem elsewhere, you're basically throwing some information away - you can't tell if both triggers are up or both triggers are down. Why would you want to do that when any software that needs to make such a weird amalgamation of two effectively unrelated values (since I think it's fair to argue that left and right triggers aren't directly related) should do it itself.

FactSphere, 2015-10-08:
Alright, alright. I see where you guys are coming from. I just wanted the OPTION to have this weird behavior enabled as a workaround for some software that expects it. Just a check box in the preference pane, that's all. If the user doesn't click it, absolutely nothing changes.

What it really comes down to is whether you have had that request made often enough to warrant it being an option for users to enable. I can't make that judgement, but you said that you have had the request a number of times. Don't you want to throw them a bone?
But if you still say no, then that's ok and I won't push any further for it.

slawo, 2015-10-08:
An option like this one will silently break behaviours that are already implemented.

It would create confusion for users and new edgeless for devs to handle.

As I already pointed out even microsoft who is king of its own OS and hardware drivers did not change a flawed behaviour (1 axis for both triggers) until they introduced a new API.

But again this project now provides a remapping tool so I guess this kind of option would not be out of the table.

fdanapfel, 2016-03-28:
Since there still seem to be quite a number of games out there that have issues when the triggers are reported as separate axes (one example where I just noticed this was "FarmingSimulator2015"), it would be great if this option could still be implemented.

daniel-jozsef, 2016-06-13:
It would be useful. I primarily play flight sims, and you REALLY NEED this in a flight sim.
I personally use ControllerMate to do this, and it's simple enough so you can do it with the free version, so it's not a huge problem, but some games don't allow you to select an axis, but rather require you to "move" the axis you want to select, and in this case a derived virtual joystick may or may not be selected...
So the driver allowing for this would be superior.

staminajim, 2016-06-18:
@daniel-jozsef Do you have any instructions on how to user Controller Mate to re-map the triggers to a single axis?  I'd also like to use the triggers for a plane rudder and need to map them to a single axis.

FranticRain, 2018-06-01:
I'm going to close this, as the development team seem to strongly resist the implementation of this feature, and it seems to have been inactive for a while.

closed 2018-06-01 by FranticRain