360Controller/360Controller #143: Thanks

jimmeijer, 2015-09-16:
Hello,

I'm 50 years, and i bouth a game ' The walking Dead' with a Logitech gamepad F710. With youre driver iT works perfect! Many Thanks, keep on the good work.
Best regards,
Jim meijet

emmanueljarri, 2019-06-23:
Definitely a great piece of software and a life saver!
Many many thanks to Drew for this amazing works!

I recently bought a controller because of this project, and even though it's not perfect (no forcefeedback working yet for me), it's definitely a game changer for all of us using macOs! 

Congratulations and many thanks!
Emmanuel Jarri

FranticRain, 2019-06-27:
I know this is like 4 years later, but I'm leaving the project soon due to personal issues with how the driver is being used by third parties. If you'd like to donate to me via PayPal as thanks for all of the work I've done over the past 5-ish years, you're more than welcome to. The stuff I learned along the way was well worth it, but I certainly won't stop you from donating. [The PayPal link is here](https://www.paypal.com/cgi-bin/webscr?cmd=_donations&business=MVEQJJ9S2EVTS&currency_code=USD&source=url)

emmanueljarri, 2019-06-28:
Just did it! Sad to see you going :-(
Regarding the issues with the third parties, **just guessing**, couldn't you change the licence or terms and conditions so it's only used by end-users and not redistributed in other projects?

FranticRain, 2019-06-28:
A lot of the code is not my original code. I stand on the shoulders of giants, is it were. And it would be a violation of the current license to use the code available under that license to make a new product with a more restrictive license. So the license is stuck the way it is.