360Controller/360Controller #144: Addition buttons on the Razer Sabertooth are not recognised

Inaughenth, 2015-09-17:
Using a Razer Sabertooth on Mac OS X 10.5 and Controllermate none of the additional buttons (those which are not present on the regular Xbox360 controller) appear to be recognised.  In the user's' manual they're labeled as: "Multi-Function Buttons (M1 / M2)" and "Multi-Function Triggers (Left: M3 / M4 ; Right: M5 / M6)".

The GUI in the systems pref pane does not include these buttons in either the test or binding tabs; although as I am using ControllerMate, I'm not really concerned about that.

As all the other functionality which is found in the regular Xbox360 controller appears to work, I'm not sure if I'm reporting a bug or asking for as yet unimplemented features.  Sorry!

Inaughenth, 2015-09-23:
If it helps, here's the info from the System Report:

  Product ID:	0x5d04
  Vendor ID:	0x24c6
  Version:	1.25
  Serial Number:	598221F5
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Razer
  Location ID:	0xfd313000 / 14
  Current Available (mA):	500
  Current Required (mA):	500

If there is a way to enable any sort of debugging from the binary releases, please let me know.  I can't compile on my system because of the USB IOKit include file changes in the current version of Xcode.

dportela, 2016-01-28:
The reason the Mac OS X system has no access to those buttons is that the controller doesn't make those buttons accessible to any operating system, period. The way the Razer Sabertooth works is that those extra buttons can be bound—by using the UI on the controller itself—to duplicate the function of other buttons on the controller. You have to do this on the controller, whether you are running Mac OS X or Windows, and if you don't do it, those buttons won't work.

FranticRain, 2016-08-09:
Closing this as unfixable.

closed 2016-08-09 by FranticRain