360Controller/360Controller #145: Logitech ChillStream gamepad not being recognized

Kombatologist, 2015-09-17:
The gamepad was recognized using the 0.14 unofficial driver, but is no longer being recognized as of the 0.15 betas. After installing the 0.15 driver and restarting, all 4 green lights on the gamepad light up (as opposed to a single light when using the 0.14 driver) and the gamepad isn't recognized at all in System Preferences > Xbox 360 Controllers.

Deph0, 2015-09-17:
Could you please provide the VendorID and productID from system repport?

Kombatologist, 2015-09-18:
Product ID: 0xc242
Vendor ID: 0x046d  (Logitech Inc.)

FranticRain, 2016-01-16:
This should be supported in the latest release. Let me know if you have any problems.

closed 2016-01-16 by FranticRain

monchote, 2016-01-16:
Confirmed that 0.15 (beta 4) now works with the Logitech ChillStream.