360Controller/360Controller #146: Rumble is completely broken (here is a fix)

larianstudios, 2015-09-18:
for (Feedback360EffectIterator effectIterator = cThis->EffectList.begin(); effectIterator != cThis->EffectList.end(); ++effectIterator)
        {
            if(((CurrentTimeUsingMach() - cThis->LastTime)*1000*1000) >= effectIterator->DiEffect.dwSamplePeriod) {
                CalcResult = effectIterator->Calc(&LeftLevel, &RightLevel);
            }
        }

1. Incorrect *1000*1000 multiplication in current code.
2. Mach time is not used in all the code, somewhere there are still CFAbsoluteTime stuff

Cheers

MaddTheSane, 2015-09-18:
If you could, please create a pull request.

RodrigoCard, 2015-09-27:
He provided this patch:
[patch.txt](https://github.com/360Controller/360Controller/files/3086/patch.txt)

Works for you @MaddTheSane ?

MaddTheSane, 2015-09-27:
Looks good on my end.

larianstudios, 2015-10-09:
Can you make a build please. The game target shipping date will be in December. I will ask for you help (if you don't mind) to give me a link to a stable release version people should use for the game.

Thanks!

RodrigoCard, 2015-11-02:
@larianstudios sorry for the delay. I will make a build soon.
The changes Apple made on El Capitan is hindering my work environment. ¬¬

larianstudios, 2015-11-02:
Is the driver compatible with El Cap btw/never had a chance to try it out? So far I've been working with AppleDTS support and I have

PS4 wired (rumble)
PS4 BT (no rumble)
PS3 wired (rumble)
PS3 BT (RUMBLE for the first time ever on OS X :)
All the XBox360 PLIST hardware (rumble when avail)
GreenAsia hardware (rumble).

I will keep you posted with game schedule ;)

RodrigoCard, 2015-11-02:
@larianstudios Yes, it is compatible. You "just" cant build it in El Cap.

I am pretty interested in getting rumble working in ps3/ps4 controllers, please keep me informed :)

slime73, 2015-11-02:
[Sorry, this is off-topic to this driver...]

@larianstudios what APIs need to be used to access rumble on a wired PS3 controller? I haven't been able to get it working when I've tried previously. Does it require a certain version of OS X?

d235j, 2015-11-02:
@RodrigoCard You can build in El Cap. You just need to download Xcode 6.4 from https://developer.apple.com/downloads, put it in e.g. /Applications/Xcode 6.4.app, and use xcode-select or the DEVELOPER_DIR environment variable to point xcrun to the proper Xcode developer path.

RodrigoCard, 2015-11-02:
@d235j Oh, right!
So the correct info is
"you cant build it on Xcode 7 with 10.11 SDK"

d235j, 2015-11-02:
@RodrigoCard Yep. You can build it just fine with Xcode 6.4 and the 10.10 SDK, even on El Capitan.

larianstudios, 2015-11-03:
@slime73 @RodrigoCard All the code will be donated to you guys at the end of dev cycle. It's pure IOKit user-space calls that work on any 10.6+ machine.  I don't know whether PS4 BT will work but we will see about it.

bestwnh, 2015-11-23:
This fix work for me.(Xbox one controller in Mac OS 10.11.1)
But if I build the controller myself, it won't found my device. If I build the feedback plugin and replace it in the 15beta3. It can found my device only if I reboot the mac with the controller plug-in. But the rumble is working!!!
So when this code can be merge into the project?

larianstudios, 2015-11-23:
@RodrigoCard any news? I am finalising stuff here. Game release very soon.

From manual:

To support Xbox 360/One controller (and similar) we suggest excellent 360Controller driver. If you like the game don't forget to drop a letter to developers or support them with donations. The page with the driver is located https://github.com/360Controller/360Controller/releases here.

RodrigoCard, 2015-11-26:
@larianstudios @bestwnh
I am catching up with my TODO list here.
I will try to make a new release soon.

bestwnh, 2015-12-01:
@RodrigoCard Thank you so much. Hope to see the new release.
Anything I can help? I'm not very good at c++, but objective-c is fine. I'm an iOS developer.
And do you know about what cause my situation？

larianstudios, 2015-12-02:
You have to check where you place the kext. Remember the feedback library will always be loaded only from /System extensions. Generally it all boils down to correct signature which you can't do because you lack necessary entitlements. Only Apple and @RodrigoCard can do this ;)

You can disable kext signature verification but that's just another story.

bestwnh, 2015-12-02:
@larianstudios I use the script to create the install mpkg, so the path should be correct.
Any specific entitlements I need? I also have a developer account to sign the project and no sign error show up. @RodrigoCard Any help?

larianstudios, 2015-12-02:
You need developer cert that is approved by Apple to sign kexts

bestwnh, 2015-12-02:
Thanks so much. I find out some resources. I do need to request the kext sign approve by apple. But I can still code. So if I can make any help just notice me.

nilbot, 2015-12-06:
Getting hyped to this new release. I'm also thinking about implement some support for the DS4 controller, because I hope that with daemon in between translating all input to a XInput-compatible scheme will actually make a lot of steam games on mac "work". Figuring out how to rumble on ds4 would be a huge win.

larianstudios, 2015-12-08:
Rumbling on PS4 with wire is easy

```objective-c
-(bool) setRumbleLarge:(AkReal32)in_fLarge small:(AkReal32)in_fSmall
{
    uint8_t uLarge = (uint8_t)(in_fLarge * 255);
    uint8_t uSmall = (uint8_t)(in_fSmall * 255);
    
    uint8_t buf[32] = {0};
    
    buf[0] = 0x05;
    buf[1] = 0xFF;
    int offset = 4;
    
    buf[offset++] = uLarge;
    buf[offset++] = uSmall;
    
    buf[offset++] = uLarge;
    buf[offset++] = uSmall;
    buf[offset++] = 0xFF; //Set Blue lamp color
    
    return kIOReturnSuccess == IOHIDDeviceSetReport(_device, kIOHIDReportTypeOutput, 0x5, buf, 32);
}
```

Wirelessly it's impossible for now.

RodrigoCard, 2015-12-30:
@larianstudios could you test it? :
https://dl.dropboxusercontent.com/u/14890209/360ControllerInstall_beta4_preview.dmg

FranticRain, 2016-01-16:
This should be fixed in the latest release.

closed 2016-01-16 by FranticRain

larianstudios, 2016-01-16:
Right-o. Can be closed. Have fun playing Divinity, which rumbles as it should now.