360Controller/360Controller #147: Horipad one does not work...

creativexp, 2015-09-19:
Just bought a officially licensed microsoft xbox one controller made by Hori. Installed v1.0.4 of your driver and it does not recognise the controller on Mac. It is being recognised by system report but not in the xbox system preference menu.

I tried the 360Controller 0.15 (beta 3) driver released by @RodrigoCard and that didn't work.

 I tried the link to a driver on dropbox as well with no luck.

FranticRain, 2015-09-19:
You must post your product and vendor IDs before we can take action. There is a guide on how to find these in the README of my repository.

creativexp, 2015-09-19:
Product id = 0x0067
Vendor id = 0x0f0d
Version 1.0.1

In my system report it appears as HORIPAD ONE.

FranticRain, 2015-09-20:
I've issued a pull request that should allow your controller to work. @RodrigoCard will let you know when there is a build with this controller added.

creativexp, 2015-09-20:
Thank you.

I would like to do a clean install of this driver. Can you tell me how to remove the drivers i installed? v1.0.4 of your driver has an uninstall script but others do not. I accidentally installed version 0.4.0 after being directed there from an old article. I also installed several versions of @RodrigoCard driver. None of these have uninstall instructions

FranticRain, 2015-09-20:
If you install the latest version of the driver, there is a part of the preference pane that will uninstall the driver. That will take care of all versions of the driver.

creativexp, 2015-09-20:
great, thanks.

closed 2015-09-22 by RodrigoCard

RodrigoCard, 2015-09-23:
@creativexp please test this one https://dl.dropboxusercontent.com/u/14890209/360ControllerInstall_0.15beta3%2BHoripadOne.dmg

creativexp, 2015-09-23:
Works Great! Appreciate it :)

PR #148 (2015-09-22): Add HoriPadOne
Resolves #147
commit 4f299253 (2015-09-22): Merge pull request #148 from FranticRain/master

Add HoriPadOne , fixes #147