360Controller/360Controller #149: Can you map to hotkeys?

nittanycolonial, 2015-09-21:
I have a program that is not in the supported list. However, the program does have hot keys (F1, F2, +, -, etc). Is it possible to still use this driver and my program together? I have the driver installed and it works fine in the sense that I can map the controller. But I would still like a way to map, for example, the left trigger to F1, or other mappings like that. I'm not clear if this program supports that.

Deph0, 2015-09-22:
The best program for this is probably ControllerMate in conjunction with this driver. It allows you to map keyboard to the controllers buttons and more.

nittanycolonial, 2015-09-25:
I looked that up and you're right, that is the next program I would need in order to implement this. Thanks!

closed 2015-09-25 by nittanycolonial