360Controller/360Controller #152: new gamespot controller

EdwardMoyse, 2015-10-08:
Hi all,

I just bought a new controller, but was sad to find it didn't work. I tried to add it to the info.plist, but being pretty rubbish with XCode, I haven't yet managed to compile yet (code-signing issues using build.sh, and  /360Controller-0.15_beta3_unofficial/WirelessGamingReceiver/WirelessGamingReceiver.h:26:10: 'IOKit/IOKit.h' file not found" if I try within XCode)

Details below:

Private Label Wired Controller for Xbox One:
  Product ID:	0x013a
  Vendor ID:	0x0e6f
  Version:	1.01
  Serial Number:	0000066DF669B24F
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14200000 / 11
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

EdwardMoyse, 2015-10-08:
After a bit more reading around, it looks like my "'IOKit/IOKit.h' file not found" is caused by my using El Capitan. Damn. It's weird because running the build.sh script works (once I removed the 'signed code' check).

d235j, 2015-10-08:
You might have the wrong SDK set. Though keep in mind, Xcode 7 with the 10.11 SDK removed the legacy IOKit USB headers. You may need to build with Xcode 6.4 and the 10.10 SDK.

EdwardMoyse, 2015-10-08:
I doubt I have space on this laptop to download Xcode 6.4 (which I assume is what you meant?)
:-(

EdwardMoyse, 2015-10-09:
Update: I tried installing 6.4 but unfortunately I just don't have enough space. :-(

I don't suppose there's any way you can build a new version with the device details above is there? I'm away from home this week and was really hoping to be able to use my new gamepad in my hotel room in the evenings, since Italian TV isn't a lot of use to me ;-)

FranticRain, 2016-01-19:
I added this controller to the driver. Let me know if it doesn't work. https://drive.google.com/file/d/0B5PLCKoMd4zGNTBKcFhodGpmU2M/view?usp=sharing

closed 2016-01-19 by FranticRain

EdwardMoyse, 2016-03-06:
Hi, sorry for the very late reply, but I just found time to try again and it doesn't seem to work unfortunately. 

The device is seen in 'system information' but in the pref pane I just see 'No devices found' (this is with Version 0.15 (beta 4) unofficial)

In console I see e.g. 
06/03/16 12:14:10.479 System Preferences[715]: No devices found? :( device count==0, i==0

I'll try with beta5 (but posting this now so I don't lose the edits when the MBP restarts)

EdwardMoyse, 2016-03-06:
And it works perfectly with beta5! Thanks!