360Controller/360Controller #154: Steam Controller support?

darktable, 2015-10-17:
The new Steam Controller is mostly broken on Macs. I'm wondering if adding it as a device to 360Controller would help it function better as a dual analog gamepad? I believe in Windows it emulates an XInput device.

[image]

Sappharad, 2015-10-17:
I actually had the same thought last night. I'm going to try it tonight (adding the controller to the driver), I'll report back if it works.

This should be seen as a temporary fix, since based on the reply I got from Valve yesterday it sounds like they want to solve this via a firmware update. If they're not making their own driver or modding this one, I'm guessing they'll add a mode to the controller so it reports as a standard USB HID controller instead of XInput.

FranticRain, 2015-10-17:
I can pretty much assure you that there is no way that this will work.

Sappharad, 2015-10-18:
I can confirm that it does not work.
I added both the Wired and Wireless Product / Vendor ID's. Confirmed that a real 360 controller was still recognized by the driver, then tried the Steam controller.

Both wired and wireless, it functions as a Keyboard/Mouse after plugging it in and Steam is not running. 360 control panel doesn't find the device. Then trying the same thing again with Steam running in Big Picture mode (so the controller works in there, but no longer as a mouse) it also isn't picked up by the driver as a 360 controller.

There may be some special command that we can send to the steam controller to "Enable" the XInput mode, but just out of the box + modified driver does not.

darktable, 2015-10-18:
Well, I really hope that Valve has hired some of the people working on this project to fix the Steam Controller support.

Sappharad, 2015-10-18:
They really don't need anything from this project to accomplish their goals. Both Windows and OS X had support for USB Game Controllers before XInput even existed. It was rumored that they just did their own thing to prevent third-party controllers from being used with the Xbox and to collect Licensing fees on third party devices. I don't know if that's true, but XInput only took over as the Xbox gained popularity.

Games on OS X that work with controllers are just asking for standard USB controllers. I believe that's all this 360 driver does, is translate XInput devices so they can be recognized as standard controllers. (Feel free to correct me.) It could save them time if they only had to support XInput, but I don't think they'll want to ask Steam users to install a Kernel Extension if they don't have to.

darktable, 2015-10-18:
From my experience with OS X games, the ones that recognize XInput controllers using this driver have all their buttons/axes mapped correctly, and ones that recognize my PS3 controller as a Bluetooth HID controller are a crapshoot that may require total remapping. So the people behind this driver have done something right.

FranticRain, 2015-10-18:
It isn't necessarily that this driver does something right. In fact, it does a lot of things wrong that we can't fix now. It succeeds because it is a driver for the industry standard controller. Developers target that controller, using this driver, for their games. They don't always test using a DS3 or DS4.

slime73, 2015-11-10:
The latest Steam Beta update includes an official installable driver for the Steam Controller on OS X, apparently.

Don't expect an Xbox controller driver to be designed to support Valve's Steam controller. ;)

Sappharad, 2015-11-10:
I noticed over the weekend that keyboard emulation was finally working for it too. Not sure exactly when it got fixed (only looked a few days back into the changelog) but based on that it seemed like it was going to be fixed very very soon. I guess it would probably be appropriate to close this, since Valve will support their own controller on OS X?

MaddTheSane, 2015-11-10:
Agreed.

Closing.

closed 2015-11-10 by MaddTheSane