360Controller/360Controller #155: Remap triggers?

pedetentous, 2015-10-17:
Hello! First off, I want to say fabulous work on this. It has prevented many headaches that I used to have with generic controllers on OS X. That being said... I'm not sure what's going on, but Mark of the Ninja seems to be re-configuring the controller when I try to play. I can remap the buttons that are incorrect, but the triggers are somehow mixed up with the analog buttons. There is no way to remap the triggers, however. Is it possible to allow binding of those for situations like this that might arise? I looked up information on the game, and it seems like there really isn't a way to remap in the game itself. Thank you so much for your time! :)

~Ped

FranticRain, 2016-01-19:
I just tested Mark of the Ninja and it works just fine with a wireless Xbox 360 controller and a wired Xbox One controller. Additionally, remapping triggers as buttons is not a great idea. You would have to go through a series of buttons to get to the one you wanted, and then precariously hold a trigger in just the right position for the perfect amount of time. Additionally, it creates so much space for user error. Like imagine a trigger with one of the buttons mapped to start. Then you would open the menu for the game every time you tried to get to a specific button. The entire remapping menu would be a complete mess, especially when Mark of the Ninja works perfectly fine.

closed 2016-01-19 by FranticRain

pedetentous, 2016-01-21:
Thank you for taking the time to check on this and respond. It turns out the version I got from a friend was a wrapped Windows version, so there could have been all sorts of issues because of that. I ended up purchasing the game on Steam, and as you said it worked just fine!