360Controller/360Controller #156: Xbox One Wireless Adapter for PC Support

mejamiewilson, 2015-10-24:
Microsoft have released the wireless dongle for Xbox One controllers (I don't know if it's compatible with 360 controllers). Support for this guy would be great, as you can have 8 controllers connected to it. 

I have bought one if anyone wants to see any info from it.

FranticRain, 2015-10-24:
We are definitely going to need to product and vendor IDs for that thing. Also, it doesn't support 360 controllers. Just Xbox One controllers.

mejamiewilson, 2015-10-24:
Here's the result of running `system_profiler -detailLevel full` on Mac OS X.

Good to know that it doesn't support 360. Is the scope of this project locked down to 360 controllers? Or is it ok to have Xbox One specific functionality?

```
XBOX ACC:

          Product ID: 0x02e6
          Vendor ID: 0x045e  (Microsoft Corporation)
          Version: 1.00
          Serial Number: 310241
          Speed: Up to 480 Mb/sec
          Manufacturer: Microsoft Inc.
          Location ID: 0x14200000 / 7
          Current Available (mA): 1000
          Current Required (mA): 500
          Extra Operating Current (mA): 0```

FranticRain, 2015-10-24:
Xbox One is welcome, I'm the guy that does all of the Xbox One work for this driver.

mejamiewilson, 2015-10-24:
Fantastic! If this issue gets to the top of the pile, let me know what I can do to help. Driver related stuff is definitely over my head.

geowar1, 2015-10-29:
FranticRain: I have a lab full of Macs, an Xbox One controller and the new wireless USB dongle for Xbox One controllers if you want/need me to test anything. You may contact me at "geowar 1 @ mac . com" (without spaces).

natbro, 2015-11-01:
i haven't yet been able to get through the XBox One dongle's initial handshake to talk through as works with XBox One wired. anybody else have ideas/leads? @FranticRain - ditto, happy to send you a dongle and/or controller if you don't have yet: natbro-at-gmail-dot-com.

geowar1, 2015-11-02:
My thought was to use a beagle (USB packet sniffer) on a windows PC and record USB traffic to the dongle 1st sans the controller and then with.

FranticRain, 2015-11-02:
You really don't even need the beagle. You can just do it with Wireshark and USBPcap, but a beagle would probably work best. If you're interested in the Wireshark method, [you can check it out here. These are instructions for sniffing the Xbox One controller itself.] (http://www.definedbehavior.com/reverse-engineering-the-xbox-one-controller/) Once you figure out the packets, its basically just an exercise in figuring out what means what and then you're good to go. Or if you don't want to figure it all out, just send me the info and I'll try and piece it together for you. I can't guarantee it'll work though. I've never looked at the 360 wireless adapter code before.

geowar1, 2015-11-02:
FranticRain has already got the 360 dongle working on OS X. We need to do the same for the Xbox One. (I'll assume that's what you meant).

FranticRain, 2015-11-02:
No, I was a little vague there. I would need to extend the 360 wireless adapter code to make it work with the Xbox One code. I've just never looked at the code this driver uses to make wireless adapters work. That's what I meant.

electric-monk, 2015-11-02:
Ahem, I did the original wireless 360 dongle support ;P

I just used the Windows driver + some USB snooper app (no idea what it was at this point, but its icon may have been Snoopy). This didn't give me any functionality the Windows driver didn't have though (and I did use a Beagle for some things, such as to deduce wired ChatPad support though.

Some people with more XBox 360s to lose than me used a Beagle on the wireless dongle inside the real console to work out the off command (and perhaps Wireless chatpad support, I think someone sent the required magic code to me to make that work) as the 360 basically just contains the same thing that the dongle does to speak to the controllers. To get XBone wireless support working nicely may also require someone to disassemble a real console and Beagle it, too.

FranticRain, 2015-11-02:
My bet is that Windows 10 is tied close enough to the Xbox One controller that disassembly will not be required. With stuff like the Xbox app, and the app to rebind controller buttons as part of Win 10, we should be able to get everything we need.

natbro, 2015-11-02:
yes, seems likely that crazy disassembly won't be required. but hopefully no irritating signed handshake or other nonsense :( i'll try to capture the new Win10<->new-dongle handshake in USBPcap on Monday and share, my home Windows machine is currently in a sad state, can't do tonight.

natbro, 2015-11-03:
too busy today to cap, hopefully asap this week.

oxycoder, 2015-11-21:
Any update on this progress?

marzman, 2015-11-21:
I have a mac running windows 10 on a virtual machine -- I have successfully connected the adapter to my windows 10 but have not been able to get my xbox one controller to pair with it -- any ideas on how to get this working ? thanks for any help

FranticRain, 2015-11-26:
@electric-monk We have a sniff of the packets going to and coming from the wireless adapter with USBPcap. You can find it over here: https://github.com/360Controller/360Controller/issues/160#issuecomment-159952216
The file works with Wireshark.

electric-monk, 2015-12-01:
I'll take a look when I find a bit of time ;)

CleanCut, 2016-01-19:
@electric-monk @FranticRain I'm willing to contribute [a dongle](http://www.amazon.com/Microsoft-Xbox-Wireless-Adapter-Windows-one/dp/B00ZB7W4QU/ref=sr_1_1?ie=UTF8&qid=1453241001&sr=8-1&keywords=xbox+one+wireless) if that would help get wireless xbox one controllers working.

On a side note, when I plug my brand-new Xbox One controller via the USB cable, it gets detected, but the keys are very wrong.  I was able to mostly remap the keys, except for the analog sticks.  Is there some trick to getting the analog sticks remapped correctly?

RodrigoCard, 2016-01-19:
please check this version @CleanCut , the mappings are fixed https://drive.google.com/file/d/0B5PLCKoMd4zGNy15NFR1TTRxVkE/view?usp=sharing

CleanCut, 2016-01-21:
@RodrigoCard That fixed the mappings, but there's other problems.  In order to not hijack this thread I created issue #207 to discuss it.

Queuecumber, 2016-02-11:
Are you guys having any luck with this? We aren't having much success on the Linux side

FranticRain, 2016-02-11:
No one has been working on it as far as I know. I haven't bothered to buy a wireless adapter yet, and @electric-monk hasn't said much of anything.

drejjmit, 2016-02-12:
Tried SnoopyPro as electric-monk suggested but nothing shows up. No data at all.
[](http://i.imgur.com/nlxm6bq.png)

FranticRain, 2016-02-12:
USBPcap + Wireshark is a way better way of capturing data. We already have a sniff though, it really just needs to be tested and implemented. From #160, https://drive.google.com/file/d/0B381VMfcesrlNWZwRkpoTXJBaDA/view?usp=sharing

drejjmit, 2016-02-12:
Did 2 caps with USBPcap and 1 with Wireshare. Maybe they will help
[https://drive.google.com/file/d/0B3-WQPnZBRZhRmlNZjltU1pWTms/view?usp=sharing](https://drive.google.com/file/d/0B3-WQPnZBRZhRmlNZjltU1pWTms/view?usp=sharing)

FranticRain, 2016-02-29:
Finally got a wireless adapter. Will take a shot at it later this week and over spring break. Word on the street is that it is encrypted. Sounds like this is going to be fun.

TyanColte, 2016-03-03:
Good Luck

maxiedaniels, 2016-03-29:
I just bought the adapter and wanted to see how the progress on this is going? Thanks!

FranticRain, 2016-03-29:
Going nowhere fast. I'm by no means well informed with the USB spec, but that whole thing is a complete mess. I can't just fake it until I make it like I did with the wired controller. I doubt I'll ever figure it out, and I'm pretty busy with school anyway.

stevenvachon, 2016-04-04:
I just bought a wireless adaptor for my Xbox One controllers as well. Disappointed that I have to use the included USB cables. Anxious for a new release!

haiduc32, 2016-04-06:
So if I was to invest some sleepless nights trying to write the driver, can anyone give me a crash course into the libraries and at what should I be looking?
@FranticRain are you positive it's encrypted? any links?

Queuecumber, 2016-04-06:
@haiduc32 As far as I know we have no idea if it's encrypted or not, it might just be complicated

FranticRain, 2016-04-06:
@haiduc32 For driver documentation, its basically nonexistent. Any that exists is poorly written and outdated. The best documentation you can get on IOKit drivers is [here.](https://developer.apple.com/library/mac/documentation/DeviceDrivers/Conceptual/IOKitFundamentals/Introduction/Introduction.html#//apple_ref/doc/uid/TP0000011-CH204-TPXREF101) Other than that, your best bet is to just study the wireless driver code currently in this driver. That is how I wrote the Xbox One wired driver. 

As for documentation on the actual packets, I recommend [my own documentation](franticrain.github.io) because, as far as I know, its the only documentation. It has information on how to capture packets from devices, the format of the Xbox One wired packet, and some various stuff I have discovered about the wireless packets.

Now, as for USB packet documentation. The driver *probably* follows the USB HID guidelines, but I can't honestly prove that. For that documentation, you are going to need to look at [the USB website.](http://www.usb.org/developers/hidpage/)

I wish you the best of luck, and if you have any questions, let me know.

(I just pushed the wireless documentation to my website, it usually takes a couple minutes to update the github.io page for no apparent reason.)

haiduc32, 2016-04-06:
Thanks, I was more interested in getting a fast understanding of the project libraries, how it works for the 360 wireless adapter so I could try to get a fast start at the One. Starting the driver from scratch doesn't sound very tempting..

FranticRain, 2016-04-06:
The only documentation I know about the wireless 360 controller adapter is the original Tattiebogle stuff. http://tattiebogle.net/index.php/ProjectRoot/Xbox360Controller/WirelessUsbInfo

haiduc32, 2016-04-06:
Thanks. Will try to sniff it tonight, see if I get any patterns. I'm concerned about the encryption, as it would be a show stopper. Will keep you posted if I make any progress.

haiduc32, 2016-04-09:
HELP! So far my experience was like going to a bar, getting wasted, then getting on a motorcycle and driving blind folded up hill through the woods. 

I got to the point where I think I know what I should be sending on the bus, but I have no idea what I'm actually sending and where.. so.. **does anyone know how to intercept USB traffic on OSX El Capitan?** It's ridiculous but I haven't found any tools..

FranticRain, 2016-04-09:
Wow, you know a lot more about these packets than I do! Awesome!

As for intercepting traffic on El Cap, there is no good way, but if you are working with the driver, the driver should let you know every time its gotten data. You can use IOLog to output that data into whatever format you want. It will show up in your "system.log" in the Console.app. I recommend adding a prefix to all of your data so you can quickly do a search in the log for your prefix and filter out all non-essential packets. That's what I'd do at least. I don't know if anyone has any better suggestions.

haiduc32, 2016-04-09:
@FranticRain thanks, but I did that already. I need to check exactly how the packets are formatted, especially the headers. I am trying to get a vm up with 10.10, people have reported that USB interception works on that version.

FranticRain, 2016-04-09:
@haiduc32 Good luck then. I'm super excited to hear you have a plan!

Queuecumber, 2016-04-09:
@haiduc32 Hey, I have a small request, when you get a chance how hard would it be for you to document some of the thing you found out w.r.t. the protocol for the wireless driver? I'm working with the Linux side of things are we are having a lot of trouble as well.

haiduc32, 2016-04-09:
@Queuecumber I'm not a "documentation" person.. I think code will be the best documentation. But let's see first if I manage to get it working, or let's have a chat if you want to try things in parallel. Drop me a private message if you want to share contact info (skype etc). 

@FranticRain thanks..

tomlube, 2016-04-10:
Anyway I can help with the wireless support?

haiduc32, 2016-04-10:
@tomlube if you have 2 controllers, I'd appreciate a sniff of the USB traffic.

FranticRain, 2016-04-11:
Turns out I only had 3 AA batteries, so in my sniffs, Controller 1 was powered from the wall, while Controller 2 was running batteries. Hope that's not a problem. I captured on Windows 8.1 using a 2015 controller (Controller 1) and a 2013 controller (Controller 2).

twoControllerInitial - Activate sniff, plug in adapter, Windows does some USB driver installing garbage like it always does, two previously paired controllers connect.
twoControllerIdle - Just an idle of the two connected controllers, started sniffing after connected
twoControllerAPress - I pressed A on Controller 1 and then A on Controller 2.

Note that these controller numbers may not be accurate, they may be swapped. I only know what I believe to be 1 and 2 since there is no LED ring.

File was zipped using 7-Zip on Windows 8.1, should be perfectly safe: [Two Controller Sniffs.zip](https://github.com/360Controller/360Controller/files/212103/Two.Controller.Sniffs.zip)

geowar1, 2016-04-11:
Use the HID Manager to register a Get Report callback and then (hex) dump everything that it receives.

haiduc32, 2016-04-17:
hey guys, wanted to drop by and say hello.. also I managed to get the wireless adapter into pairing mode (led blinking) but the SoB is stubborn and doesn't want to pair with anything.. any way, I got 2000 lines of code just to get to this stage, and my eyes hurt.. I'll have time only next weekend to continue and will release some code once I manage to pair with the controller so other people can try it out if it works with their adapters.

haiduc32, 2016-04-25:
hi guys, I need a few more sniffs, as for pairing I need a specially constructed packet, which is different in 3rd party scans and I still haven't figured out exactly how to build it.
so far I got sniffs from FranticRain and drejjmit, more sniffs especially while pairing, will be highly appreciated!

tomlube, 2016-04-25:
Hey Haiduc, how could I sniff these for you? Do you need multiple controllers/adapters/Windows/OSX/what are the requirements and how could I help?

FranticRain, 2016-04-25:
@tomlube This should get you on the right track for how to sniff. Sniffs need to be on Windows. OS X ones won't help.

looterz, 2016-04-28:
@haiduc32 Here is a sniff I ran on a USB Hub with only a keyboard and the wireless adapter connected. It contains packets of the adapter searching for a device, finding the device and pairing, then about 5-10 seconds of traffic after a successful pair.

**OS**: Windows 10 x64
**Devices**: XBox Wireless Adapter for Windows (v21.50.45.656), XBox One Controller (v6.2.11059.0)
**pcap**: https://db.tt/36tawx5e

haiduc32, 2016-04-28:
Thansk, I've seen the linux thread and plan to use their sniffs for checking as well. I am kind of busy with life at the moment and not sure how much time will be able to afford on spending on this project, so things might go slower. I'll try to post my current work so if anyone is available, to be able to chip in.

tomlube, 2016-04-28:
OS: Windows 10 Pro x64
Devices Xbox One Wireless adapter for Windows/Xbox One Controller
USBPcap: http://www35.zippyshare.com/v/Lz32B1fS/file.html

rpcastagna, 2016-04-29:
Hey @haiduc32 if you start a repo/branch with the code you have so far I might be able to help get it a little further; I've been pretty busy with life stuff too but I spent long enough pulling these packets apart that I feel obligated to try to make some meaningful contribution out of it, haha. I'll post the captures I'd made while playing with it, also.

rpcastagna, 2016-04-29:
Various captures: 
OS: Windows 10 Student x64
https://www.dropbox.com/s/fwvqdj3cdz7hdcj/Xbox%20One%20Wireless%20Caps.zip?dl=0

haiduc32, 2016-05-01:
Hi, so this is my fork: https://github.com/haiduc32/360Controller
I'd appreciate if someone could try it out (I mean developers that know how to run it) and check if their wireless adapters are entering into pairing mode (led blinking). It does for me.
For those that want/have time to contribute to the effort, the class is OneWirelessGamingReceiver. There are checkpoint comments with line numbers that correspond to sniffs from dejjmit: https://drive.google.com/file/d/0B3-WQPnZBRZhRmlNZjltU1pWTms/view?usp=sharing the 1.pcap
I started a doc for my investigation: https://drive.google.com/file/d/0B3-WQPnZBRZhRmlNZjltU1pWTms/view?usp=sharing if anyone wants to contribute let me know and I'll provide read&write permissions.

FranticRain, 2016-05-01:
I went ahead and made a signed build of this so people can try it out without having to go through the trouble of building it. [You can get that here.](https://drive.google.com/file/d/0B5PLCKoMd4zGbmNtVTdENVN0ZzQ/view?usp=sharing) **PLEASE NOTE THAT THIS WILL NOT MAKE YOUR WIRELESS CONTROLLER WORK AND IS HIGHLY UNSTABLE. IF YOU COMPLAIN ABOUT THIS NOT WORKING I WILL COME TO YOUR HOUSE AND SNAP ALL OF YOUR XBOX ONE CONTROLLERS IN HALF,**

I got my wireless adapter into pairing mode by pressing the button on the adapter. This caused the light to blink. I then took a 2013 controller and put it into a sync mode. 10-20 seconds later, I received a kernel panic. I didn't expect the controller to pair, but I definitely didn't expect a kernel panic.

haiduc32, 2016-05-01:
@FranticRain thanks for trying it out. don't know why it panicked, did not happen for me. maybe highlight the **RISK OF KERNEL PANIC!!!** so people are trying it on their own risk.

FranticRain, 2016-05-01:
@haiduc32 It's important to note that I still kernel panic whether or not I activate the pairing mode. Based on your console outputs, its doing an awful lot of idle talking, and then it panics. I can't get anything specific out of it, but I definitely won't be messing around with it for a couple weeks. Hooray for finals.

[75 more replies]

PR #935 (2019-10-14): Add wireless dongle support
It's finally ready to be released. All input functionality should be working just fine, although button/stick remapping and deadzones are not implemented yet. Some serious problems that need to be fixed:

- [x] ~macOS crashes when shutting down (kernel panic in IOHIDFamily)~ *known issue*
- [x] ~Controller doesn't work in most games (though it works fine in control panel)~ *not fixable*
- [x] ~Serial number is always blank in IO Registry~ *fixed*

~Any ideas/hints are more than welcome~ Most issues have been fixed 😄.

Closes #156
closed 2019-10-14 by FranticRain