360Controller/360Controller #157: Metro 2033 Redux Button Mapping Incorrect

IVXCVI, 2015-10-26:
The buttons are all over the board. What can I do to troubleshoot this?

erisler-myntpartners, 2015-12-08:
bump.

Running Yosemite + 360 Wireless Controller.

B seems to behave as if it's A.
Right trigger, normally fire, causes a "look right" movement. 
Back is start,
etc.

I've tried version 14 through 15 beta 3 with the same result.
I've tried the disable feature of the driver but it seems to turn off the controller and then Metro doesn't recognize the controller. If I turn it back on, then the preferences pane seems to react to the controller but the "enable driver checkbox is still unchecked.

Any suggestions?

FranticRain, 2016-01-24:
You can try disabling the driver from within the preference pane just in case the game attempts to subvert the driver, but there isn't much we can do to fix game specific issues. We can do our best to create a generic interface that developers hook in to, but there is no way for us to tell the driver: "Hey Metro just started, change the mappings." If it doesn't work, you can try to take it up with the developers, but based on the Steam forums for this game it looks like they don't care at all.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain