360Controller/360Controller #158: Assassin's Creed 2: Controller not working

Promthanius, 2015-10-26:
Hey guys,
I bought AC2 from the AP store, got one of those xbox 360 wired controllers because the WASD and mouse was too annoying to play with. I got the driver thing, and paid for Joystick Mapper. But the game still doesn't know the controller is there and makes me play WASD. Please help. What am I doing wrong? yes, I leave Joystick Mapper open when playing.

GB-CG, 2015-10-28:
@Promthanius You could try Enjoy2. https://github.com/fyhuang/enjoy2 Although, some may have 360 issues with that too, but give it a go.

RodrigoCard, 2015-11-02:
This game itself is not compatible with game controllers.
Also, Joystick Mapper, Enjoy and other similar software uses this driver, but its problems are not directly related. So I am closing this issue.

If you need support for other software, please contact its authors. :)

closed 2015-11-02 by RodrigoCard