360Controller/360Controller #159: Hori Ex2 Fightstick not found

Copas27, 2015-10-28:
This build is a lot better than the original 360 driver, however even though the plugin recognizes that there is a controller plugged in, my games don't respond or acknowledge its presence.
[image]
[image]
^ normally skullgirls displays a controller connected. 
[image]
^my ps3 controller works but not my 360 controller.

RodrigoCard, 2015-11-02:
This device's IDs are not on the list. It should be added in a future update, probably.

FranticRain, 2016-01-16:
This device should be supported now. Let me know if you have any issues.

closed 2016-01-16 by FranticRain