360Controller/360Controller #160: Xbox Elite Controller not working yet.

chadthman, 2015-10-28:
Just got the Elite Controller and unfortunately it appears to not be supported yet. Here is what it shows in system profiler if that helps:

``` 
Controller:

  Product ID:	0x02e3
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.03
  Serial Number: <12 digit alpha-numeric number here>
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x1a153000 / 11
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

TyanColte, 2015-11-18:
Same Problem here, also the standard controller with the latest firmware update is also not working. Please fix. I miss my controller.

FranticRain, 2015-11-21:
Forgot to mention it since I added it before this issue was created (October 3rd) but this is resolved in #151.

TyanColte, 2015-11-21:
@FranticRain do we have a release to go with this resolution?

FranticRain, 2015-11-21:
I don't do releases, but since the pull request hasn't been accepted, there it is not in any current build. But as far as I know, we can't build because the build system is broken (see #163) and the driver is broken in the latest version of El Capitan (see #161 and #167.) As far as I can guess, the maintainers are waiting for Apple to stop screwing around with the driver API and stop breaking things.

TyanColte, 2015-11-21:
@FranticRain well that's a bummer. When do you speculate a new release will be? I'm going to be going on a trip soon and I really would like all this to work. :+1:

FranticRain, 2015-11-21:
Unfortunately, I can't speak for the rest of the dev crew. They get it done when things work and they have the time. If you're willing to manually install the driver, I can compile a bootleg version of the driver. I never have figured out how all of the build scripts fit together, but I do have a kext signing certificate. Let me know if you are interested.

TyanColte, 2015-11-22:
@FranticRain sure, i'm up for testing it. let me know when you've got one for me. I'll go ahead and plop it in and let you know how it worked.

FranticRain, 2015-11-22:
Okay, download the version @herrbischoff was nice enough to compile [here.](https://github.com/360Controller/360Controller/pull/151#issuecomment-158693976)

Then move the "360Controller.kext" into `/Library/Extensions`
You can get to this directory easily by using the `Go -> Go to Folder...` or `CMD + Shift + G` option in Finder.
Once it is there, click on it, and then make an alias in Finder with `File -> Make Alias` or `CMD + L`.
Move the Alias to `/System/Library/Extensions` and put it there.
Once that is placed, you should be good to go. You'll probably need to reboot before it'll work.

TyanColte, 2015-11-22:
@FranticRain Awesome! I'll let you know if it works.

FranticRain, 2015-11-22:
Ah, he didn't sign the kext. That's why its not working. I should have a kext signing certificate, but I recently reinstalled OS X and appear to have lost it. I'll do my best to recover it tomorrow.

TyanColte, 2015-11-22:
@FranticRain I have a friend who may be able to accomplish this for me. Thanks for the help you've provided so far.

TyanColte, 2015-11-23:
@FranticRain so i was finally able to get this installed, I had to allow kext-dev-mode=1 and disable System Integrity Protection on my entire mac. But it works, it loads the kext. It also recognizes the elite controller, but the controls are all wonky, moving a control stick around pushes random buttons and directions on the D-Pad, among other issues (maybe having to do with the ability to remap any control as a different control on your Xbox One or windows 10). It also still doesn't recognize the original Xbox one controller with the latest firmware update installed on it.

FranticRain, 2015-11-23:
I swear if Microsoft changed the packet layouts on me again, so help me. I'll get Windows 10 set up and update my controller. Then I'll sniff the packets to see whats up.

TyanColte, 2015-11-23:
Knowing M$ that wouldn't surprise me. They really don't want mac users using this stuff on their macs. Even though it's the best of all the controllers on the market.

TyanColte, 2015-11-23:
@FranticRain Do you all have an IRC or anything I can get in to help contribute with. Chatting in comments like this is kind of an odd duck.

FranticRain, 2015-11-23:
What is the product and vendor ID on your original Xbox One controller? That thing should be working on the latest version of the driver unless you have a third party controller.
As for chatting with comments, its just the typical fair for this kind of this. We don't have any secret dev chat or anything.

TyanColte, 2015-11-23:
Lol, ok, i referenced the product and vendor IDs of my original xbone controller in #151 latest comment.

TyanColte, 2015-11-23:
@FranticRain if you want, i can set up wireshark and usbpcap on my windows 10 machine and cap some packets for you. i'll send you the results of you send me some way i can do so.

FranticRain, 2015-11-23:
Yes! If you could do that, it would be incredibly useful. Here's how:
Follow the "Following Along" section of [this website](http://www.definedbehavior.com/reverse-engineering-the-xbox-one-controller/). Once you are done there, skip down the to "The stuff magic is made of" section and sniff the packets. If you could just send me the whole capture files, that would be great. Additionally, if you could make one scan of you pressing buttons in a particular order on the controller, that would be very useful. The order the buttons used to be in was A, B, X, Y, LB, RB, View, Menu, Left Stick Click, Right Stick Click, Guide, Sync, Up, Down, Left, Right, Left Stick, Right Stick, Left Trigger, Right Trigger.

FranticRain, 2015-11-23:
And if for some reason you are interested in looking at my notes for previous packet layouts, you can see those [here](https://www.dropbox.com/s/25dd5a6irn7jfqt/RAWPACKETS.html?dl=0).

TyanColte, 2015-11-23:
@FranticRain Got the cap, can't upload it here. Will put it on my GDrive and send you a link.

TyanColte, 2015-11-23:
@FranticRain https://drive.google.com/file/d/0B381VMfcesrldnFuOE1aSmpDREE/view?usp=sharing

FranticRain, 2015-11-23:
@TyanColte According to that, the only packets being set from the computer to the controller is an init packet that is already in the driver. `05:20:10:01:05` and `05:20:0f:01:05` from your scan are similar to the sent driver's `05:20:03:01:00`. You'll notice the 3rd byte differs. That is just an increment byte that isn't important. The last byte is different though. I've never seen a trailing `05` before. Which controller was this?

TyanColte, 2015-11-23:
That was the version 1 xbox one controller with the latest firmware update received from the xbox accessories app in windows 10. it's the controller without the 3.5 mm jack on the bottom. nothing else is connected to the controller, not a chat pad or anything.

FranticRain, 2015-11-23:
Okay, can you modify your testing a little bit? Open up USBPcap with the controller plugged in. When it gives you the filters, unplug the controller. Select the filter that the controller was in, give it a name, and then plug the controller in. Give it a couple seconds and then quit scanning. Then let me see that one.

TyanColte, 2015-11-23:
Do i need to do that using wireshark or just usbpcapcmd by itself?

FranticRain, 2015-11-23:
Just USBPcapCMD. No need for Wireshark. You really only need Wireshark if you want to look at the data.

TyanColte, 2015-11-23:
ok, will do. i'll get you a report asap

TyanColte, 2015-11-23:
@FranticRain https://drive.google.com/file/d/0B381VMfcesrlVzZuaGZ3Q2tYbXc/view?usp=sharing is this what you needed?

FranticRain, 2015-11-23:
Yeah. That's it, and its virtually identical to the existing sniffed packets. I don't see anything wrong here.

TyanColte, 2015-11-23:
Don't forget i'm using that unsigned version of the driver that @herrbischoff compiled. all i know is that when i plug it in it vibrates to show that it's getting power and then nothing.

TyanColte, 2015-11-23:
but even when i was using the beta 3 driver it still wasn't working

FranticRain, 2015-11-23:
The controller vibrates when using the OS X driver? It should just light up the Xbox logo. Its not supposed to vibrate. It should only produce rumble on Windows.

TyanColte, 2015-11-23:
It does vibrate, as soon as i plug it in. I don't know to what end it does this for. However the elite controller does the same thing but then it lights up.

FranticRain, 2015-11-23:
The Elite controller vibrates and then lights up on OS X?

TyanColte, 2015-11-23:
Yea, with that one driver that @herrbischoff compiled.

TyanColte, 2015-11-23:
Could it have something to do with the fact that i was using the xone driver before this? i removed the kext after uninstalling the entry in system preferences. but i may have some left over files.

FranticRain, 2015-11-23:
Xone-OSX only has two files. The Pref Pane and `Xone Controller.kext`

TyanColte, 2015-11-23:
well those have been removed. so there's that. hmmmmmm. I'm going to try reinstalling the 360controller package and see if that helps. I'll uninstall the driver first. as long as i can get the original xbox one controller working, i will be ok until the elite controller becomes supported.

FranticRain, 2015-11-23:
The 2013 Xbox One controller works with the driver as far as I know. I'm paranoid about updating my 2015 controller because I still want to play Fallout 4 on my Windows 8 machine. >_>

TyanColte, 2015-11-23:
what about the 2013 xbox one controller with the latest firmware update? because that's what i have and it wasn't working.

FranticRain, 2015-11-23:
I don't have any 2013 controllers with me right now. I'll try and update my 2015 one and see if it has problems too.

TyanColte, 2015-11-23:
are you using your xbox one to update your controller?

FranticRain, 2015-11-23:
No. My Xbox One is not with me at the moment. I plan to update through Windows 10.

TyanColte, 2015-11-23:
Oh, well that's about the only other way you can update it. I just reinstalled the 0.15-B3 package and rebooted my mac and still the controller isn't recognized.

FranticRain, 2015-11-23:
Just updated my 2015 controller. It still works perfectly with OS X. There is a slight vibration now, but it runs just fine.

TyanColte, 2015-11-23:
So what's the deal with mine? what am i doing wrong?

FranticRain, 2015-11-23:
I honestly have no idea. Everything looks good to me.

TyanColte, 2015-11-23:
would it matter what cable i'm using?

FranticRain, 2015-11-23:
Yes, but only if you are using a data cable vs a charge cable. If you are getting vibration and it shows up in the system profiler, you are using a data cable.

TyanColte, 2015-11-23:
Argh, when do you think you'll be able to get your hands on a 2013 controller, update it and see if it has the same problems as mine?

FranticRain, 2015-11-23:
Not for a while. It would be after Thanksgiving.

TyanColte, 2015-11-23:
Darn, well that's no good. I've gotta think of some other way. If only i could downgrade the firmware on my 2013 controller. I don't use the chat pad with it or audio or anything. I just updated it because i like to have up-to-date stuff and since i don't own an Xbone, when they released the xbox accessories app on windows 10 i was like "w00t" and then i was like "crap..."

FranticRain, 2015-11-23:
Okay, lets just test this from scratch again, just to make sure.
Get your 2013 controller. Remove all accessories from the controller. Remove all batteries from the controller. Before plugging the controller in, run `kextstat | grep -i 360Controller`. This command shouldn't output anything.

Open the console. You can access this from `Applications -> Utilities -> Console`. Then in the left hand panel, click on system.log. Make note of what the last message you see is. We will be interested in the output after we connect the controller.

Now plug your controller in, observer and note its behavior. Now run `kextstat | grep -i 360Controller` again. This time it should have some sort of output. Something like this:
```
141    0 0xffffff7f832f9000 0x9000     0x9000     com.mice.driver.Xbox360Controller (1.0.0d15) 9DF80CD4-09D9-3E65-8CA0-AE3D42E0D206 <39 31 5 4 3>
```
If you don't see that, tell me what the last items listed are when you run the `kextstat` command by itself. I don't need the full output, just the last couple. Additionally, if the kextstat grep doesn't return anything, check and make sure you can see the controller in the system profiler `About This Mac -> System Profiler -> USB -> Controller`

Check the output of the console for any message prefixed by `start - `. If you see anything like that, post the recent contents of your system.log.

TyanColte, 2015-11-23:
@FranticRain I'll do just that as soon as I get home from work. About 6:00 American Central Time. So about 6.5 hours from this posting.

TyanColte, 2015-11-24:
@FranticRain Ok, so i did everything you said when i plugged in my controller i got two weird things in my console system.log

    Nov 23 18:21:49 Christophers-MacBook-Pro kernel[0]: start - unable to open device
    Nov 23 18:21:49 Christophers-MacBook-Pro cfprefsd[136]: BUG in libdispatch: 15B42 - 1631 - 0x0

and when i ran the `kextstat | -i grep 360controller` after plugging in the controller i received this output:

    148    0 0xffffff7f82da7000 0x9000     0x9000     com.mice.driver.Xbox360Controller (1.0.0d15) 9DF80CD4-09D9-3E65-8CA0-AE3D42E0D206 <43 32 5 4 3>

let me know what you think.

TyanColte, 2015-11-24:
@FranticRain i don't think the `BUG in lib dispatch: 15B42 - 1631 - 0x0` has anything to do with it because i see similar things farther up in the log but it was the two instant things that came up when I plugged in the controller.

TyanColte, 2015-11-24:
OK, something different happened that time in the log when i plugged in my controller

    Nov 23 18:40:07 Christophers-MacBook-Pro kernel[0]: start - unable to open device
    Nov 23 18:40:08 Christophers-MacBook-Pro kernel[0]: kIOMessageServiceIsRequestingClose

that second message came up

TyanColte, 2015-11-24:
And this came up in the log when i unplugged it.

    Nov 23 18:40:32 Christophers-MacBook-Pro kernel[0]: kIOMessageServiceIsTerminated
    Nov 23 18:40:32 Christophers-MacBook-Pro kernel[0]: Stopping
    Nov 23 18:40:32 Christophers-MacBook-Pro kernel[0]: Free!

FranticRain, 2015-11-24:
So what this means is that the driver matched to your controller, and started loading. But it failed to open the IOUSBDevice, which is a weird thing to have happen. The "start - " stuff is really the only part that matters in these logs. The rest of this data is typical. I'll keep looking into this, but I don't for the life of me know why the open would fail.

[55 more replies]

closed 2016-01-16 by FranticRain