360Controller/360Controller #161: Not working on OS X 11.2

hugomingoia, 2015-10-28:
The Xbox 360 controller is no longer detected on OS X 11.2 beta.

GuamBomb, 2015-10-31:
Can confirm this.  Appears that all the kernel extensions are loaded, and System Information sees the controller.  360Daemon is running.  Perhaps in temporarily disabling wireless controllers, you disabled the product and vendor IDs of my wired controller?  It's a 3rd party controller, not Microsoft, so that could be an issue.  That said, no issues on OS X 11.1 or prior.

fent, 2015-10-31:
0.14 works for me on 10.11

Also, there a couple other issues that are the same or similar #123 #127

GuamBomb, 2015-11-01:
That was true for me as well, we're talking about an upcoming update, 10.11.2.  

That said, I've also attempted a clean install with SIP disabled.  No change.

herrbischoff, 2015-11-01:
I can confirm this as well. Since installing the 10.11.2 beta yesterday, my 360 Controller is not recognized anymore. I'm using the original Microsoft USB Receiver.

Watching the console log, I noticed these relevant lines:

```
01.11.15 12:22:08,102 com.apple.xpc.launchd[1]: (com.mice.360Daemon[980]) Trampoline was terminated before jumping to service: Killed: 9
01.11.15 12:22:08,000 kernel[0]: Kext com.mice.driver.Xbox360Controller not found for unload request.
01.11.15 12:22:08,000 kernel[0]: Kext com.mice.driver.Xbox360Controller not found for unload request.
01.11.15 12:22:08,000 kernel[0]: Kext com.mice.driver.Wireless360Controller not found for unload request.
01.11.15 12:22:27,024 com.apple.xpc.launchd[1]: (com.mice.360Daemon) This service is defined to be constantly running and is inherently inefficient.
01.11.15 12:24:23,785 com.apple.xpc.launchd[1]: (com.mice.360Daemon[710]) Trampoline was terminated before jumping to service: Killed: 9
01.11.15 12:24:23,000 kernel[0]: Kext com.mice.driver.Xbox360Controller not found for unload request.
01.11.15 12:24:23,000 kernel[0]: Kext com.mice.driver.Xbox360Controller not found for unload request.
01.11.15 12:24:42,710 com.apple.xpc.launchd[1]: (com.mice.360Daemon) This service is defined to be constantly running and is inherently inefficient.
01.11.15 12:27:55,000 kernel[0]: Got serial number: 559BCBE0Un-authenticated device attached
01.11.15 12:28:42,000 kernel[0]: Un-authenticated device attached
01.11.15 12:28:49,000 kernel[0]: Got serial number: 559BCBE0Un-authenticated device attached
```

Maybe this helps someone more knowledgeable to track down the source of the issue.

GuamBomb, 2015-11-01:
Herrbischoff, some of that should probably go in a feedback to Apple. Especially the HID errors. That could be effecting just about any USB input device, and might be the reason I'm getting less than perfect response from my Logitech wireless keyboard/trackpad device. Keyboard works okay, but the trackpad response occasionally lags for no apparent reason.

herrbischoff, 2015-11-01:
@GuamBomb I have already opened a Radar issue with the relevant information. Let's see if and how Apple responds. I have a feeling that this may be related to the change from IOHIDDevice to IOUSBHostHIDDevice in the 10.11 SDK.

herrbischoff, 2015-11-02:
I have completely cleaned up my machine and tried running the driver again. Here's some more console output when starting up the preference pane:

```
02.11.15 15:52:30,641 System Preferences[768]: Failed to connect (bindingInputColumn) outlet from (Pref360ControlPref) to (NSTableColumn): missing setter or instance variable
02.11.15 15:52:30,641 System Preferences[768]: Failed to connect (linearDeadzoneLeft) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
02.11.15 15:52:30,641 System Preferences[768]: Failed to connect (linearDeadzoneRight) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
02.11.15 15:52:30,642 System Preferences[768]: Failed to connect (xoneRumbleOptions) outlet from (Pref360ControlPref) to (NSPopUpButton): missing setter or instance variable
02.11.15 15:52:30,644 System Preferences[768]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fc1666560e0 H:[MyWhole360Controller:0x7fc166652690(465)]>",
    "<NSLayoutConstraint:0x7fc166656960 H:|-(NSSpace(20))-[MyWhole360Controller:0x7fc166652690]   (Names: '|':NSView:0x7fc163c1ab90 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc16669ff60 h=-&- v=-&- H:|-(0)-[NSView:0x7fc163c1ab90]   (Names: '|':NSTabView:0x7fc163c1ae80 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc1666a0630 h=-&- v=-&- H:[NSView:0x7fc163c1ab90]-(0)-|   (Names: '|':NSTabView:0x7fc163c1ae80 )>",
    "<NSLayoutConstraint:0x7fc166631c10 H:[NSTabView:0x7fc163c1ae80]-(0)-|   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166631c60 H:|-(0)-[NSTabView:0x7fc163c1ae80]   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166656db0 H:[MyWhole360Controller:0x7fc166652690]-(NSSpace(8))-[NSBox:0x7fc166649bf0'Left Stick']>",
    "<NSLayoutConstraint:0x7fc16664ce00 H:[NSBox:0x7fc166649bf0'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x7fc163c1ab90 )>",
    "<NSLayoutConstraint:0x7fc16664c030 H:[NSBox:0x7fc166649bf0'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fc163c1ba70 H:[NSBox:0x7fc166631070'Box']-(0)-|   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166631800 H:|-(0)-[NSBox:0x7fc166631070'Box']   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc1666308a0 H:[NSBox:0x7fc166631070'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fc1666560e0 H:[MyWhole360Controller:0x7fc166652690(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
02.11.15 15:52:30,657 System Preferences[768]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fc1666560e0 H:[MyWhole360Controller:0x7fc166652690(465)]>",
    "<NSLayoutConstraint:0x7fc166656960 H:|-(NSSpace(20))-[MyWhole360Controller:0x7fc166652690]   (Names: '|':NSView:0x7fc163c1ab90 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc16669ff60 h=-&- v=-&- H:|-(0)-[NSView:0x7fc163c1ab90]   (Names: '|':NSTabView:0x7fc163c1ae80 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc1666a0630 h=-&- v=-&- H:[NSView:0x7fc163c1ab90]-(0)-|   (Names: '|':NSTabView:0x7fc163c1ae80 )>",
    "<NSLayoutConstraint:0x7fc166631c10 H:[NSTabView:0x7fc163c1ae80]-(0)-|   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166631c60 H:|-(0)-[NSTabView:0x7fc163c1ae80]   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166656db0 H:[MyWhole360Controller:0x7fc166652690]-(NSSpace(8))-[NSBox:0x7fc166649bf0'Left Stick']>",
    "<NSLayoutConstraint:0x7fc16664ce00 H:[NSBox:0x7fc166649bf0'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x7fc163c1ab90 )>",
    "<NSLayoutConstraint:0x7fc16664c030 H:[NSBox:0x7fc166649bf0'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fc163c1ba70 H:[NSBox:0x7fc166631070'Box']-(0)-|   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166631800 H:|-(0)-[NSBox:0x7fc166631070'Box']   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc1666308a0 H:[NSBox:0x7fc166631070'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fc1666560e0 H:[MyWhole360Controller:0x7fc166652690(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
02.11.15 15:52:30,978 System Preferences[768]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fc1666560e0 H:[MyWhole360Controller:0x7fc166652690(465)]>",
    "<NSLayoutConstraint:0x7fc166656960 H:|-(NSSpace(20))-[MyWhole360Controller:0x7fc166652690]   (Names: '|':NSView:0x7fc163c1ab90 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc16669ff60 h=-&- v=-&- H:|-(0)-[NSView:0x7fc163c1ab90]   (Names: '|':NSTabView:0x7fc163c1ae80 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fc1666a0630 h=-&- v=-&- H:[NSView:0x7fc163c1ab90]-(0)-|   (Names: '|':NSTabView:0x7fc163c1ae80 )>",
    "<NSLayoutConstraint:0x7fc166631c10 H:[NSTabView:0x7fc163c1ae80]-(0)-|   (Names: '|':NSView:0x7fc163c1b320 )>",
    "<NSLayoutConstraint:0x7fc166631c60 H:|-(0)-[NSTabView:0x7fc163c1ae80]   (Names: '|':NSView:0x7fc163c1b320 )>",
[... cut ...]
Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fc1666560e0 H:[MyWhole360Controller:0x7fc166652690(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
02.11.15 15:52:30,981 System Preferences[768]: No devices found? :( device count==0, i==0
02.11.15 15:52:31,062 System Preferences[768]: Result of kextstat = 256
```

Spidey002, 2015-11-02:
Hope someone figures this out. Until then, I'm going back to the original release.

herrbischoff, 2015-11-02:
Yeah, hopefully. If I had *any* background in Objective C, I would donate my time but unfortunately I do not. Let me know if there is anything else I can do to help along resolving this.

Aeonofgods, 2015-11-02:
I can confirm this as well. My Xbox One controller does not work anymore

Spidey002, 2015-11-02:
I reverted back to Mac OS 10.11.1, and the controller works again. I'm gonna hold off on the beta until hopefully someone can engineer a solution. 

Reverting wasn't too hard, just reinstalled El Capitan from the App Store and then did NOT update to beta.

Aeonofgods, 2015-11-02:
@Spidey002 So all you had to do was go into purchases, el capitan, and download? How can you opt out of the beta so it doesn't update on it's own?

Spidey002, 2015-11-02:
I don't think I have auto updates turned on.

Aeonofgods, 2015-11-02:
Ok but all you did was click download on El Capitan? And all your files and everything remained the same correct?

Spidey002, 2015-11-02:
@Aeonofgods Yes! I went to my purchases tab and clicked download on El Capitan.

Aeonofgods, 2015-11-02:
@Spidey002 awesome thank you very much!

RamblinLamb, 2015-11-02:
So I already had 10.11.1 installed prior to installing the game pad driver.
Said driver at 0.15 doesn't see my Xbox 360 game pad....

-- 
Richard T. Lamb
*Ramblin Lamb Photography*
35905 SE Boulder Ct
Snoqualmie WA 98065
rtlamb@ramblinlamb.com
Http://www.ramblinlamb.com <http://www.ramblinlamb.com/>
B - 425.427.2573
C - 425.444.6243
F - 425.831.6840

Aeonofgods, 2015-11-02:
It is worth a shot for me because this driver and a different one I had been using were working up until this update

Spidey002, 2015-11-03:
@Aeonofgods Do make sure you turn off automatic updates under System Preferences > App Store. 
Mine was somehow turned back on, and when I started up this morning, I was running the beta again. :/

Aeonofgods, 2015-11-03:
Ok I will make sure to do that. Thanks for the warning

herrbischoff, 2015-11-03:
Just reverted to 10.11.1 and it's working like a charm again.

Apple did respond with a request for system profiler save files. If people who encounter the same problem that I have could send me theirs, I will submit them all directly to Apple.

To avoid having them posted here for posterity, send to me via email. The address is in my profile.

To obtain a full System Information Report (also known as ASP, system profile, spx, system configuration):

1. Select About This Mac from the Apple Menu
2. Click the System Report button on the About This Mac window
3. Select Save from the File menu (cmd-s)
4. Change the file name and save location as needed
5. Click the Save button

This report can also be obtained in Terminal using the following command:

```
/usr/sbin/system_profiler -detailLevel full -xml > ~/Desktop/mymachine.spx
```

mr-mcox, 2015-11-04:
@herrbischoff I'm seeing almost identical console output while running 10.1.1. I'll email you my system profiler save in case its helpful.

herrbischoff, 2015-11-04:
@mr-mcox Thanks for your email. I have attached this in my bug report to Apple.

herrbischoff, 2015-11-09:
Update: Apple engineering just responded.

> Engineering has determined that your bug report is a duplicate of another issue and will be closed. 

The original bug report was closed as well which either means it got dropped or fixed. Sorry for not being able to give more information. Apple's bug reporting tool is rather opaque.

d235j, 2015-11-09:
@herrbischoff: what was the ID number of the bug you opened?

herrbischoff, 2015-11-09:
@d235j: 23347588

d235j, 2015-11-10:
@herrbischoff Beta 3 just came out; I suggest trying it and seeing if this is fixed.

herrbischoff, 2015-11-10:
@d235j: Will do so immediately and report back.

herrbischoff, 2015-11-10:
@d235j: It's almost working. The wireless controller is found but no button input appears to be recognized.

Log output:

```
10/11/15 22:38:23,743 System Preferences[609]: No devices found? :( device count==0, i==0
10/11/15 22:38:23,752 System Preferences[609]: No devices found? :( device count==0, i==0
10/11/15 22:38:28,214 System Preferences[609]: No devices found? :( device count==0, i==0
10/11/15 22:38:32,013 System Preferences[609]: No devices found? :( device count==0, i==0
10/11/15 22:38:35,772 System Preferences[609]: No devices found? :( device count==0, i==0
10/11/15 22:38:40,287 System Preferences[609]: No settings found
10/11/15 22:39:08,287 System Preferences[609]: Unable to simultaneously satisfy constraints:
(
    "<NSLayoutConstraint:0x7fb52a57c730 H:[MyWhole360Controller:0x7fb52a674c90(465)]>",
    "<NSLayoutConstraint:0x7fb52a52c1e0 H:|-(NSSpace(20))-[MyWhole360Controller:0x7fb52a674c90]   (Names: '|':NSView:0x7fb52b89dc20 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fb52b9523e0 h=-&- v=-&- H:|-(0)-[NSView:0x7fb52b89dc20]   (Names: '|':NSTabView:0x7fb52b89df50 )>",
    "<NSAutoresizingMaskLayoutConstraint:0x7fb52b952430 h=-&- v=-&- H:[NSView:0x7fb52b89dc20]-(0)-|   (Names: '|':NSTabView:0x7fb52b89df50 )>",
    "<NSLayoutConstraint:0x7fb52b8a4480 H:[NSTabView:0x7fb52b89df50]-(0)-|   (Names: '|':NSView:0x7fb52b89e3f0 )>",
    "<NSLayoutConstraint:0x7fb52b8a44d0 H:|-(0)-[NSTabView:0x7fb52b89df50]   (Names: '|':NSView:0x7fb52b89e3f0 )>",
    "<NSLayoutConstraint:0x7fb52a5a02f0 H:[MyWhole360Controller:0x7fb52a674c90]-(NSSpace(8))-[NSBox:0x7fb52b8cb280'Left Stick']>",
    "<NSLayoutConstraint:0x7fb52b8cde30 H:[NSBox:0x7fb52b8cb280'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x7fb52b89dc20 )>",
    "<NSLayoutConstraint:0x7fb52b8cd690 H:[NSBox:0x7fb52b8cb280'Left Stick'(145)]>",
    "<NSLayoutConstraint:0x7fb52b8a3f10 H:[NSBox:0x7fb52b8a0710'Box']-(0)-|   (Names: '|':NSView:0x7fb52b89e3f0 )>",
    "<NSLayoutConstraint:0x7fb52b8a3fb0 H:|-(0)-[NSBox:0x7fb52b8a0710'Box']   (Names: '|':NSView:0x7fb52b89e3f0 )>",
    "<NSLayoutConstraint:0x7fb52b8a2a70 H:[NSBox:0x7fb52b8a0710'Box'(668)]>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x7fb52a57c730 H:[MyWhole360Controller:0x7fb52a674c90(465)]>

Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
```

galad87, 2015-11-11:
It started working for me on 10.11.2 beta 3.

GuamBomb, 2015-11-17:
I reverted back to 10.11.1 and opted out of beta updates for the time being.  With the fourth beta out, I decided to risk a try.  Working nicely so far.

RodrigoCard, 2015-11-21:
apparently it was fixed on latest beta. Anybody else?
I think we can close this.

herrbischoff, 2015-11-21:
According to my Apple bug report:

> The original report on the issue, Bug ID 23321900, was closed and the resolution is “fixed in OS X 10.11.2”.

So it seems Apple has confirmed the bug and its resolution in the upcoming 10.11.2 update. This issue should be closed as soon as the official update is out and confirmed working. For the time being I think leaving this issue open may help others seeking for help with the same problem.

quanganhdo, 2015-12-09:
10.11.2 is officially released today. Can anyone confirm if the driver works or not? 

Thanks.

FranticRain, 2015-12-09:
I just tested it using an Xbox One controller, and it looks like it works just fine. I didn't get any rumble out of Fez though, so thats probably a bug. But just using it as a controller seems perfectly fine.

sbouafif, 2015-12-09:
Same for me.

There was no problem since Beta3 of 10.11.2 I think and it still works well since the last update

lbonfante, 2015-12-30:
Can someone confirm if the OSX 10.11.2 Final version (15C50) works fine and with out problem with the "Drivers360Controller Version 0.15 (beta 3)", i will buy the xbox360 controller, please where can i download the .dmg installer?

FranticRain, 2015-12-30:
The driver works perfectly fine in 10.11.2. Closing.

closed 2015-12-30 by FranticRain

lbonfante, 2015-12-31:
Thanks, i bought the control and it works excelent and very well with OSX 10.11.2 and "Drivers360Controller Version 0.15 beta 3" :) so happy, let's play!!!