360Controller/360Controller #162: 360 Wireless Receiver not recongized (OS X 10.11.1)

3breadt, 2015-10-31:
Hi,

I cannot get my 360 Wireless Receiver working with the latest driver (0.15 beta 3) under OS X 10.11.1. It won't allow the controller to connect.

The console gives the following error:
```
31.10.15 11:30:15,000 kernel[0]: 011396.946043 IOUSBHostDevice@04100000: IOUSBHostDevice::start: device descriptor request completed with 0xe00002ed and bytesTransferred 0
31.10.15 11:30:15,000 kernel[0]: 011396.948007 PRT1@04100000: AppleUSBOHCIPort::resetAndCreateDevice: failed to start device
31.10.15 11:30:15,000 kernel[0]: 011396.948015 PRT1@04100000: AppleUSBOHCIPort::resetAndCreateDevice: failed to create device (0xe00002e9), disabling port
31.10.15 11:30:15,509 UserEventAgent[45]: assertion failed: 15B42: com.apple.telemetry + 26475 [B2764968-61C9-31B7-8063-9CDC792388CB]: 0xffffffffe00002c7
31.10.15 11:30:15,523 System Preferences[1453]: No devices found? :( device count==0, i==0
31.10.15 11:30:15,000 kernel[0]: 011397.765875 IOUSBHostDevice@04100000: IOUSBHostDevice::setConfigurationGated: compliance violation: failed to enable remote wake
31.10.15 11:30:16,033 System Preferences[1453]: No devices found? :( device count==0, i==0
```

The wireless receiver is not the official one from Microsoft but a chinese clone. Under Windows it only works when you force the official MS driver to be used for it, as I think it has a wrong USB PID (0x0291) compared to the official receiver.

Does your driver not support cloned receivers or is it an issue specific to El Capitan?

RodrigoCard, 2015-11-02:
it is probably because of the IDs, I think.
Could you give us the Product AND the Device IDs?

3breadt, 2015-11-02:
Vendor ID: 0x045E
Product ID: 0x0291
Revision: 0x0107

akv12, 2015-11-06:
mine still isn't working.. which it won't work on my macmini.. but it works on my macbook!

3breadt, 2015-11-06:
My issue was probably related to the receiver itself being broken. I tried it on my PC with Windows 10 and could not get it work.

I now tested a new receiver from Microsoft and it works like a charm. So it is not an El Capitan related issue, but just a broken receiver in my case.

closed 2015-11-06 by 3breadt

indie1982, 2016-08-12:
I know this is old but I have exactly the same issue, cheap Chinese clone receiver from eBay with the same IDs and revision as above but I know it definitely works, it's fine on RetroPi using xboxdrv.

It just won't connect to the controller when it's connected either of the MacBook Pros (OS X 10.11.6, XBox 360 Controller Driver 0.16.3) in the house.

Any ideas on what I can do to troubleshoot?

indie1982, 2016-09-04:
@RodrigoCard Sorry to bump this again, but I have a wireless receiver with the same IDs as @3breadt. It works when I plug it into my Raspberry Pi and use it with RetroPi and xboxdrv but when I plug it into my MacBook Pro and use these drivers it doesn't work. I can't get the receiver and controller to pair.

I've updated to 0.16.4 just in case that fixed it but nope :(

FranticRain, 2016-09-04:
I cannot recreate this issue. I'm running the same version of macOS with the same version of the driver and my wireless receiver with that VID and PID works just fine with the driver.

indie1982, 2016-09-04:
@FranticRain Thanks for the reply, I'm glad yours works but mine doesn't! Is your a chinese clone or an official receiver with the same IDs?

What data can I give you that would help? Does the driver manage the connection between the receiver and the controller or should that work independently of the driver?

I'm pretty sure I'm not doing anything wrong in the connection phase of the setup: Plug the receiver in to the computer, make sure the light on it is flashing, if it isn't press the button on it to make it flash. Power on the Xbox 360 controller and press the connect button on the rear whilst the receiver is also flashing? I guess at this point they should just pair and the light on the controller go solid?

FranticRain, 2016-09-04:
It is a Chinese knock-off, just like yours. I don't know what data you can collect and what you need. The Wireless 360 code is not my domain. You'll have to wait for someone else to help you. I just wanted to qualify that this is a user-specific issue, not something fundamentally broken with the Wireless 360 implementation.

indie1982, 2016-09-04:
Is there anyone in particular that I can tag? I asked a few weeks ago and got no reply, I guess because the issue in the thread is closed.

It's strange that I can't get it to work on two different models of MacBook Pro and not just my main machine.

FranticRain, 2016-09-04:
No. There isn't anyone in particular. Someone might show up at some point when they feel like working on the driver and fix it for you. I don't know. None of us get paid to do this, and we never talk outside of issues, so its just sheer luck if anything gets fixed.

electric-monk, 2016-09-04:
The pairing/etc. happens entirely within the dongle, with some on-board firmware the host computer has very little control over. From what I recall, it'll even work if the device isn't configured on the USB bus (e.g. is only receiving power).

If it works on Linux with xboxdrv then there's certainly the chance there's just something missing, but who knows what. From what I know both Linux drivers are based on the info on my site, though (xpad used to specifically thank me regarding the wireless info, though apparently someone deleted that at some point :P ), so I suppose it might be possible to just compare what the code's doing. I don't have time just now though :(