360Controller/360Controller #163: Local build in Xcode 7 fails

herrbischoff, 2015-11-01:
On OS X 10.11.1 and 10.11.2 beta with Xcode 7.1 the `./build.sh` script fails with the following error message:

```
** BUILD FAILED **

The following build commands failed:
	CompileC build/360\ Driver.build/Release/360Controller.build/Objects-normal/x86_64/ChatPad.o 360Controller/ChatPad.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
	CompileC build/360\ Driver.build/Release/360Controller.build/Objects-normal/x86_64/Controller.o 360Controller/Controller.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
(2 failures)
******** BUILD FAILED ********
```

The relevant lines in the compiler log seem to be:

```
In file included from /Users/[REDACTED]/360Controller/360Controller/ChatPad.cpp:30:
/Users/[REDACTED]/360Controller/360Controller/_60Controller.h:27:10: fatal error: 'IOKit/usb/IOUSBDevice.h' file not found
#include <IOKit/usb/IOUSBDevice.h>
```

and 

```
In file included from /Users/[REDACTED]/360Controller/360Controller/_60Controller.cpp:27:
/Users/[REDACTED]/360Controller/360Controller/_60Controller.h:27:10: fatal error: 'IOKit/usb/IOUSBDevice.h' file not found
#include <IOKit/usb/IOUSBDevice.h>
```

MaddTheSane, 2015-11-01:
This is in part due to Apple changing the driver model used in 10.11.

If needed, you can get Yosemite installed in a Virtual Machine and use Xcode 6.

d235j, 2015-11-21:
@MaddTheSane You don't need a virtual machine — just download Xcode 6.4, copy it somewhere, and perhaps rename it.

The Mac OS 10.11 SDK changed the USB driver API, as Apple rewrote the USB stack.

herrbischoff, 2016-02-23:
As a significant rewrite appears to be necessary to make compilation work on the current generation of Xcode, I'm closing this for the time being. My question has been answered.

closed 2016-02-23 by herrbischoff