360Controller/360Controller #164: Support Findway® Wired USB Controller

hatpick, 2015-11-05:
I used your driver to work with one of these gamepads to play PES 2016, it seems that it has all the keys wrong in that game, is there any way that I can verify that the driver is working fine with this gamepad?

RodrigoCard, 2015-11-09:
This is a windows game played through a compatibility layer, right?
the button ids of this driver doesnt match the windows' mappings.

Try to remap the buttons inside the app itself.
If that is not possible, try the ramapping in the "binding" tab inside the Xbox Preference pane.

hatpick, 2015-11-09:
Thanks for the help, it is not possible to remap inside the app, where can I find this xbox preference pane?

FranticRain, 2015-11-09:
Apple Menu -> System Preferences. There will be an icon near the bottom that looks like an Xbox 360 controller. That is the one you want. One of the tabs within that area will let you rebind the buttons on the controller.

hatpick, 2015-11-09:
Found it, I'll report the results here.

FranticRain, 2016-01-19:
Assuming resolved due to lack of response.

closed 2016-01-19 by FranticRain