360Controller/360Controller #167: El capitan

akv12, 2015-11-06:
I installed Beta 15 3 on my macbook and my mac mini.. my controller which is wireless works.. with the wireless receiver on my macbook. However, I have uninstalled all the drivers and everything and reinstalled the drivers and rebooted and it still won't work on el capitan on my mac mini.. Can someone please help me get this to work. I been looking for hours. Please help!

akv12, 2015-11-06:
Please someone help me!

akv12, 2015-11-06:
Also can someone make an UN-installation  script or app so Its easier to uninstall all the drivers! Please? I really want to play my game I just bought..now I feel like I wasted money!

mazedlx, 2015-11-06:
Driver isn't working anymore with OS X El Capitan 10.11.2 Beta (15C27e)

akv12, 2015-11-06:
Hi, how did u get to work before?

mazedlx, 2015-11-06:
before 0.14 worked fine. upgraded recently to newest public beta, since then: nothing. upgraded the driver to 0.15b3 and still nothing

akv12, 2015-11-06:
.14 works but .15 never worked you are saying?

mazedlx, 2015-11-06:
no.
0.14 worked on the previous el capitan public beta (10.11.1) but not on 10.11.2 beta, as well as 0.15

akv12, 2015-11-06:
I can't even get the controller to work. at all on my mac.. it works on my macbook but not my macmini and i am so annoyed

akv12, 2015-11-06:
Have you tried to uninstall your drivers completely??

mazedlx, 2015-11-06:
yes, to no avail.

akv12, 2015-11-06:
how do I uninstall it?? I am having trouble doing it.. maybe i am doing something wrong.. can you help me?

mazedlx, 2015-11-06:
i just removed the control panel icon with a right click

akv12, 2015-11-06:
well that isn't enough... you aren't fully uninstalling... go to this link:http://forums.macrumors.com/threads/uninstall-xbox-360-controller-driver.447793/

mazedlx, 2015-11-06:
thanks, will try that und reinstall 0.15b3

akv12, 2015-11-06:
tell me if you have luck! I need help myself and I am annoyed!

herrbischoff, 2015-11-08:
Please see https://github.com/360Controller/360Controller/issues/161

There is currently no way to get this to work. You need to revert back to 10.11.1, then the driver starts working again.

mazedlx, 2015-11-12:
Happy to say that all issues went away with the current beta release (10.11.2 Beta (15C40a))

akv12, 2015-11-12:
how I have it and i am having trouble!

mazedlx, 2015-11-12:
uninstalled every trace of the driver. upgraded to the latest public beta, installe 0.15 beta 3 driver. done.

daslicht, 2016-01-19:
Any news regarding 10.11.x ?

RodrigoCard, 2016-01-19:
@daslicht : as @mazedlx it is working already

closed 2016-01-19 by RodrigoCard