360Controller/360Controller #168: Controller Not Recognized

scarmichael, 2015-11-07:
I have two Xbox 360 Controllers, both are the same model number, though they differ in that one is a direct USB connection, and one has the little adapter dongle to adapt the proprietary Xbox controller connector to standard USB.

With both the original Tattiebogle drivers and now the latest from this project, I am able to get my machine to recognize the controller with the dongle adapter... but not the direct USB one.

On connect, the USB-direct controller will show a green power light... but the System Pref panel never recognizes it. When the adapter-connected controller is unplugged, the driver still does not recognize the USB-direct controller, so it's not as if there is a conflict or issue with precedence.

The System Profiler does report that the USB 3.0 bus sees both controller devices.

The functional controller has the following info:

Controller:

  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.14
  Serial Number:	15B5DD1
  Speed:	Up to 12 Mb/sec
  Manufacturer:	©Microsoft Corporation
  Location ID:	0x14200000 / 13
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

The non-functional one, however:

  Product ID:	0x02a0
  Vendor ID:	0x1430  (RedOctane)
  Version:	1.00
  Serial Number:	00A36703
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft Corporation.
  Location ID:	0x14100000 / 14
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

So, clearly something is different internally on the two devices.

Any suggestions? Sorry for the wall of text.

FranticRain, 2015-11-08:
Okay, first things first. You have official Microsoft controllers. One is a wired controller. One is wireless. The cable you are talking about is part of the Xbox 360 play and charge kit. The cable is often referred to as a charge cable. It does not transfer data. No matter what cable you attach to the wireless controller, no matter what configuration, will NEVER turn a wireless 360 controller into a wired one. That is a feature of the Xbox One controller, but it does not apply to the Xbox 360 controller. You will need a wireless dongle to use the wireless Xbox 360 controller with your Mac.

Now, you claim that your machine recognizes the wireless controller, but not the wired one. But your next paragraph says that the controller lights up, and you can get a system profiler report for it. Which one is it? And when you say a "green power light," do you mean the ring around the Xbox button? What pattern is it flashing?

d235j, 2015-11-08:
@FranticRain that's one possibility.

Another possibility is that @scarmichael has one controller that's a real Xbox 360 controller. It has the disconnect adapter dongle, which is designed to prevent one from dragging the entire Xbox on the floor if the controller is yanked on.
The other controller, while looking like a real Xbox 360 controller (does it?), identifies as a RedOctane controller. This one must not be in the Info.plist, and therefore is not supported by the driver.

electric-monk, 2015-11-08:
It sounds like the "little dongle" is the little circular DIN-style connector MS put on the 360 controller as well as things like the DDR dancemats, between the USB connector and the controller. It sounds like the other is a third party controller?

FranticRain, 2015-11-08:
I think you guys are right about the connector, that makes sense. Would have helped if he was a little more clear. As for RedOctane, I think they only made plastic guitars for the Xbox 360, not an official controller. He does claim that both controllers are the same model number though, which makes me think they are both Microsoft products.

scarmichael, 2015-11-08:
Gents, sorry if I was not clear. 

Both are wired Xbox 360 controllers. OEM as far as I know (or the second on me was the most believable fake I have ever seen). 

Controller 1 (we will call it) has the DIN style connector that connects then via USB. This is the one that IDs as Microsoft as the vendor in the system profiler. It talks to the driver just fine. Functions just fine. 

Controller 2 is direct USB. No DIN connector. It is not wireless, just USB-only. It IDs in the system profiler with Red Octane as the vendor and, yes, I too am aware of their normal product line. The box was all MS branded though. This controller when plugged in will flash green around the center logo button, just a single full circle flash, and then nothing else. Is not recognized by the driver, but is recognized in the system profiler as a connected device. 

Both devices are printed with the same model number on the back. I can see no sign of one device being a knock-off. 

Hopefully that's clearer. 

Controller

RamblinLamb, 2015-11-08:
This was the mistake I was making! I "assumed" that the cable was both 
for charging and data. I was wrong. Once I paired my controller to the 
wireless receiver the driver/controller worked!

I had no idea the cable was for charging only.

Thank you for helping clear the air!

Now if I can get Dirt3 to stop crashing upon startup life will be much 
better....
-- 
Richard Lamb
425.427.2573
rtlamb@ramblinlamb.com<http://www.getpostbox.com>

FranticRain, 2015-11-08:
@RamblinLamb Was it a circular cable that looks sort of like S-Video? Or was it more of a rectangular brick with a light that attached to the top of the controller?

FranticRain, 2015-11-08:
I added the vendor and product id to #151 so whenever that gets accepted (in the next decade most likely) this will probably be fixed.

RamblinLamb, 2015-11-08:
It's grey/thin/round cable with a white connector that connects to the
front of the controller above the battery. This apparently doesn't carry
any data, who knew? Not me. I've had a wireless dongle but wasn't going to
use it until after I got the other cable/controller to work, which we now
know will never work.

An interesting lesson learned for sure.

-- 
Richard T. Lamb
*Ramblin Lamb Photography*
35905 SE Boulder Ct
Snoqualmie WA 98065
rtlamb@ramblinlamb.com
Http://www.ramblinlamb.com <http://www.ramblinlamb.com/>
B - 425.427.2573
C - 425.444.6243
F - 425.831.6840

d235j, 2015-11-08:
@FranticRain can you revert that addition, in this case? It won't help us if this is just a charge cable.

scarmichael, 2015-11-08:
So mine (original poster) is a hard-wired USB controller. RamblinLamb's issue is entirely separate. 

Thanks for the vendor add. Hopefully that solves things.

FranticRain, 2015-11-08:
@d235j Its been my experience that charge cables do not appear in the system profiler.

d235j, 2015-11-08:
Oh, sorry, I got confused here. @FranticRain is right.

electric-monk, 2015-11-08:
@FranticRain in the past it certainly appeared, I think as product code +1 of the wired controller. It appeared to need some configuration to actually initiate charging too, though I never looked into it. However, I did use the ID to pop up an alert telling the user that they needed a wireless controller, which I did to hugely reduce the number of e-mails I got about people trying to use the charging cable with the driver (after that I'd just get the occasional e-mail from people who didn't read the alert)

FranticRain, 2015-11-08:
@electric-monk Interesting. My experience has mostly been with charge vs data micro USB cables, so these proprietary things might be different. I guess the ultimate solution is:

@scarmichael If you plug in the breakaway controller into a Windows PC or Xbox 360, does it work?

scarmichael, 2015-11-09:
@FranticRain So, the breakaway unit is the one that is working fine with my Mac using your driver (and the Tattiebogle one before it)

The direct-USB unit is the one that doesn't connect. I'm going to test that tomorrow on a Windows PC once I have access to one. I suspect it's going to work, but I'll report back in.

scarmichael, 2015-11-09:
@FranticRain Works immediately when associated with a Windows VM. So, the device is functional. Here's hoping the vendor addition will take care of things.

FranticRain, 2015-11-09:
So in conclusion, the breakaway controller is official Microsoft, it works. Pure wired controller is third party, even though it looks official, and it doesn't work. I've added the VID and PID of the third party controller in a pull request. When that is built, this problem will be resolved.

scarmichael, 2015-11-09:
@FranticRain That is correct. Arrived sealed in a MS branded box, but vendor in the System Profiler shows third party. Thanks so much for your help.

electric-monk, 2015-11-09:
I'm beginning to think it might be worth getting pictures of the thing, as that's really bizarre!

scarmichael, 2015-11-09:
@electric-monk I actually took these yesterday in case I needed to prove I wasn't dealing with a wireless controller.

http://imgur.com/a/YxsQW

The left one in the top image is the functional controller (black product info label), the right one (white product info label) is the one that isn't working on my Mac. Everything about them physically appears to be official MS.

electric-monk, 2015-11-09:
Well, those things do certainly look similar - the only real "tells" I could see that the non-Microsoft one isn't Microsoft is the lack of a ferrite bead, a cheap QA sticker and the fact that the words "XBox 360" and "Controller" somehow don't manage to line up on the sticker, which I can't imagine Microsoft would have done.

Is it possible this is a weirdly well done knockoff? The people making it may have used RedOctane's firmware, or just plain Vendor ID, to save getting their own.

scarmichael, 2015-11-09:
They feel pretty similar. I played a bit of the RedOctane one today on Mighty Gunvolt, a little Mega-Man-esque platformer, and it felt just fine. Behaved as I would have expected.

I found this image online of another controller, and the label matches up exactly except for the spacing you noted on "Xbox 360 Controller": http://i.imgur.com/t4pB0oOl.jpg

The hologram sticker is legit and the stamping on Xbox 360 is as well, per this page: https://www.microsoft.com/en-us/howtotell/Hardware.aspx#Xbox

I'll see if I still have the box it came in and give that a rundown as well. Now it's a mystery to solve.

electric-monk, 2015-11-10:
It's worth noting that the hologram on the RedOctane unit doesn't look right either - the "Microsoft" text isn't their logo, it's just the word "Microsoft" in a strange font, the spacing of everything seems a bit off and it doesn't even have the much smaller markings that the "good" one does.

FranticRain, 2015-12-19:
Resolved in #151

closed 2015-12-19 by FranticRain