360Controller/360Controller #169: Buttons and D-Pad intermittently don't work, or get stuck on (Xbox One with Cable)

jeshuamj2, 2015-11-10:
Sometimes the buttons or d-pad don't respond to presses or they respond and get stuck on until the sticks or the triggers are pressed. Also these seem to occasionally not work when the triggers are fully depressed, but they do work when its partially depressed.

FranticRain, 2015-11-10:
Does the controller and cable work properly with a Windows PC or Xbox One? It is important that it includes the cable, not wirelessly connected to an Xbox One.

jeshuamj2, 2015-11-23:
The controller works fine with Windows (I dual booted the Mac). It seems as if the buttons just get stuck on, and until I move a stick or a trigger, it stays on for some reason. The game I play is F1 2013, and Feral has a driver in the game, but it doesn't pick up the triggers very cleanly, whereas this driver does that very well when I enable it. But the stuck buttons is more of a handicap than the triggers are

jeshuamj2, 2015-11-23:
BTW, I'm running 10.11.0 right now. I'll try updating to 10.11.2 and see what happens.

FranticRain, 2015-11-23:
If the only game you are experiencing this with is F1 2013, its probably some sort of conflict between Feral's implementation and ours. We unfortunately can't make per-game changes for stuff like that. Or maybe you need to stop driving Caterhams and Marussias and start driving a Merc! :smile:

jeshuamj2, 2015-11-24:
Well, its the only game I test in but the issue is apparent even in the Pref Pane. 
Here's a link to a video of the preference pane as I click the buttons and they get stuck:
https://drive.google.com/file/d/0B94XokhYd0FeeEdjdTZaakxrcHc/view?usp=sharing
Haha, McLaren is personally my favorite but its not very good in this game :P

FranticRain, 2015-11-24:
This sounds a lot like it's the fault of the controller here. I'm not sure the driver is at fault. Controllers send a bunch of data whenever buttons, triggers, or sticks are moved. In your case, it looks like the controller is sending a packet when the "x" button is depressed, but it is missing the release of that button. It works properly when the trigger is depressed because your slight variations in movement of the trigger are firing lots of data packets from the controller to the driver, causing the controller to pick up on the button changes quickly.
It is of course possible that the driver is dropping packets, but there are plenty of people who don't experience this problem, so I'm not sure about that. My best guess for helping you fix this problem would be: Try using a different USB cable to connect the controller and see if it changes anything. It probably won't, but it's worth a try.

jeshuamj2, 2015-11-29:
I tried a different cable and no dice. I don't doubt that you know far more about this than I do, but if it were down to the controller, why is the issue not present when I use Feral's built-in driver, or when I use the Windows built-in driver? It works absolutely perfectly in Windows, and aside from a light trigger irregularity (which this driver fixes) in Feral's it works there too. 
Also, most times I depress the button, it doesn't pick up either. It only picks anything up when one of the triggers or sticks is moved.

FranticRain, 2015-11-29:
All of your arguments are also valid for everyone else who uses any Xbox One controller with the driver without this problem. I can play games perfectly fine without this issue, so how can it be the driver? It is something specific about your situation, I don't know what it is, and I have no idea how to figure out what the problem is. But its not something I can just magically go in and fix.

jeshuamj2, 2015-11-30:
Ah, I found the error. It seems I had one of your old drivers (xone-OSX) installed and not properly removed. I removed that and reinstalled this one and it works perfectly. 
Sorry for the trouble.

closed 2015-11-30 by jeshuamj2

FranticRain, 2015-11-30:
I'm glad you fixed it! Awesome.