360Controller/360Controller #172: Support for Speedlink XEOX Pro (SL-6566-BK)?

antazivkovic, 2015-11-23:
https://github.com/xboxdrv/xboxdrv/issues/77

Hello. It appears the guys managed to make this one work on Linux. Just follow the link.  Basically, the driver recognises the gamepad but it will not connect to the dongle. It will connect to the dongle in d-mode, but then the diver doesn't recognise it. Anz suggestions?

Thanks

Ante

nilsbecker, 2016-03-26:
this may be related: i got a speedlink strike nx wireless to pair by pushing the start button while the device was in the pairing phase. vendor id 0x045e product id 0x028e, so this may be similar hardware.

russdamerell, 2016-06-06:
Also having problems with a Speedlink XEOX Pro (SL-6566-BK) wireless gamepad. It shows up as "Xbox 230 Wired Controlled (Wired)" in the pref panel when switched in Xinput mode, but pressing buttons all moving the joysticks do nothing, no changes can be seen in the test pane or in Steam test. Any ideas?

FranticRain, 2016-06-06:
We do not support controllers that do not work with Xbox series consoles, sorry.

russdamerell, 2016-06-06:
Isn't it? (TheSpeedlink XEOX Pro)

FranticRain, 2016-06-06:
Can you get it to work with an Xbox 360 console? I highly doubt it. It imitates the VID and PID of the official Xbox 360 controller, making it a knock-off. And not only that, it doesn't follow the Xbox controller packet spec, otherwise it would work by default.

russdamerell, 2016-06-06:
Well, thaks for the info. Highly disappointed at this.

FranticRain, 2016-08-09:
Closing this, since this is an unsupported type of controller.

closed 2016-08-09 by FranticRain