360Controller/360Controller #173: Thrustmaster Ferrari Vibration GT Cockpit 458 Italia Edition

X1l10on88, 2015-11-23:
Hello, your driver works great with a standard USB Xbox 360 Controller, but this device is not recognized,  (in Windows it works great, so the device is fine), i think this device should be supported as it recognized by Windows as a standard Xbox 360 controller (and it's meant to be like this, this is stated in the driver section of Thrustmaster).

Link to the device: http://www.thrustmaster.com/it_IT/prodotti/ferrari-vibration-gt-cockpit-458-italia-edition

Thanks

X1l10on88, 2015-11-23:
OS: Mac OSX 10.9.5

TyanColte, 2015-11-23:
@X1l10on88 If you go to the apple menu and "About this Mac" and click system report at the bottom of that window, Click usb on the left and look for "Controller" it should say what the product id and vendor id, that's what they'll need to add support for that controller in the driver.

X1l10on88, 2015-11-23:
Ferrari 458 Racing Wheel:

  Product ID:	0x5b03
  Vendor ID:	0x24c6
  Version:	 1.01
  Serial Number:	00000C74
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Thrustmaster
  Location ID:	0x14200000 / 8
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

antazivkovic, 2015-11-28:
SL-6566:

  Product ID:	0x0809
  Vendor ID:	0x1a34
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14400000 / 10
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

Also a new device.

FranticRain, 2016-01-16:
The Ferrari 458 Racing Wheel should be in the latest release. Let me know if it doesn't work.

closed 2016-01-16 by FranticRain