360Controller/360Controller #174: Mac App Store support

sstrong2, 2015-11-26:
Is there any way to get this driver to work with a Mac App Store submission? Is I/O Kit supported in sandboxed apps?

TyanColte, 2015-11-26:
@sstrong2 As cool as that would be I highly doubt that'll happen. I don't think apple will allow the things that are done here to be done from an app in the app store. This is better suited for a pkg install anyway. However it would be cool if it could be done.

sstrong2, 2015-11-26:
Ok, but not sure why you say my app would be better installed from a pkg. I understand the 360controller software being installed from a pkg but not Mac Apps in general. Distributing Mac Apps through the Apple App Store has worked in the past for me. Apart from other platforms like Steam, how else would you get wide distribution of Mac Apps?

slime73, 2015-11-26:
I think your question was interpreted as 'can this driver be put on the Mac App Store?'

If your question is actually 'can my Mac App Store app use the Apple IOKit HID and ForceFeedback APIs?', then I believe the answer is yes.

sstrong2, 2015-11-26:
ok, thanks - sorry for the ambiguous question. So how can I get my Mac App Store application to work with the 360controller. Running directly on the app in my dev tool (Unity3D) it works fine. Just not when I build it for the store.

RodrigoCard, 2016-01-19:
do you still have this issue with the latest version?

sstrong2, 2016-01-20:
Haven't checked since late November. Why, has there been a recent patch to fix this is issue?

larianstudios, 2016-01-25:
Sandboxed app can use the driver absolutely fine given you grant USB permissions

sstrong2, 2016-01-25:
Which version of the driver are you using and what did your entitlements file look like?

larianstudios, 2016-01-25:
Latest 0.15.4 from FranticRain

Game log: HID Controller Connected: Xbox 360 Wired Controller
<dict>
	<key>com.apple.security.app-sandbox</key>
	<true/>
	<key>com.apple.security.device.microphone</key>
	<true/>
	<key>com.apple.security.device.usb</key>
	<true/>
	<key>com.apple.security.files.bookmarks.app-scope</key>
	<true/>
	<key>com.apple.security.network.client</key>
	<true/>
	<key>com.apple.security.network.server</key>
	<true/>
	<key>com.apple.security.temporary-exception.iokit-user-client-class</key>
	<array>
		<string>AppleMultitouchDeviceUserClient</string>
	</array>
</dict>

sstrong2, 2016-01-26:
I've tried the following entitlements file but I get no xbox controller response from my app built with Unity3D (latest 5.3.1p4). It works fine in the Unity editor but not as a app deployed on the Mac.

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
<key>com.apple.security.app-sandbox</key><true/>
<key>com.apple.security.security.device.usb</key><true/>
<key>com.apple.security.temporary-exception.iokit-user-client-class</key><true/>
</dict>
</plist>
```

larianstudios, 2016-01-26:
I am using our own Engine so I directly open HID devices..

FranticRain, 2016-08-09:
Closing this since it is impossible, and inactive.

closed 2016-08-09 by FranticRain