360Controller/360Controller #175: Both Xbox 360 and Xbox One controllers return same Product ID

pbhogan, 2015-11-29:
Both seem to show up as ProductID = 0x028e which is the Xbox 360's ID, instead of, I believe, 0x02d1 or 0x02dd for the Xbox One IDs.

Even though the mappings are the same, this matters for apps that make decisions for UI and such based on VID + PID.

FranticRain, 2015-11-29:
This is by design so that the Xbox One controller has maximum compatibility with games. If it had a different product ID, many games would ignore it and then we would have people complaining about that. Its a lose lose situation.

pbhogan, 2015-11-30:
Hmm. Seems like this could cause more problems than it solves in the long run as games do add the Xbox One to their supported list. None of the other controllers supported by this driver masquerades as a 360 controller, why should the One be different?

Maybe a better approach is an option in the pref panel to enable this 360 masquerading per controller? Have it default to off. That way it's user decision if they want/need to turn it on and it's obvious to the user why the controller is showing up "incorrectly" in UI or elsewhere, and no-one can complain either way. :)

FranticRain, 2015-11-30:
I'll look into this. I'll need to figure out how to reload the driver on command though, since this will be required for this to work. I'm currently trying to get my signing cert back from Apple, but I'll build a version of this for you to test when I get that figured out.

electric-monk, 2015-12-01:
I do like the idea of the Xbone controller appearing as its own thing, for the reasons @pbhogan states, though I can also see why you might want it to pretend to be a 360 controller. There's a few games that actually hardcode checking for my original class names (Bioshock for example), which complicates things even more. Other than that, I'd say the packet layout should reflect the native hardware format.

With regards to the driver reloading, I think that should be pretty simple, it's just a matter of a parent/child arrangement with the parent swapping the desired child in dynamically? I can't quite remember what the hierarchy actually looks like right now though...

FranticRain, 2015-12-01:
@electric-monk Oh, thats a way better idea than I had. I guess I'll get back to futzing with HID report descriptors. Yaaaaaaaaaaaay /s.

electric-monk, 2015-12-01:
@FranticRain Are you manually twiddling the bits or using the terrible Windows app the USB people provide? It's worth using even though it's terrible, I used it to make the original 360 descriptor, as well as some descriptors for USB microcontrollers I was playing with.

FranticRain, 2015-12-01:
@electric-monk I use the windows app, because if I don't, I end up programming with comments and then I can't figure out why things don't work. Its hilarious every time. I can probably just go through my old driver and rip out the old HID report descriptor and use that. That's probably the best way. I'll need you guys to take a look over it to make sure its not a complete mess since I'm not great at this kind of stuff. I'll throw it all into a pull request at some point.

electric-monk, 2015-12-01:
@FranticRain ah, that's good, I just wanted to be sure you weren't torturing yourself ;)

RodrigoCard, 2015-12-04:
About driver reloading: It should unload it properly first to work, right?
Currently it is not happening, see #66

electric-monk, 2015-12-04:
If what I suggested works (I believe it should) the driver will never actually unload, so that shouldn't be a concern.

FranticRain, 2015-12-04:
It looks like just using `padDisconnect()` and `padConnect()` works. I just had to add some more condition handling to `padConnect()`.

My changes are basically done. I need to do some testing on the Elite controller, and there is no way in the preference pane to make the Xbox One controller act like an 360 controller. Unfortunately, I haven't gotten a new kext signing cert yet, so I can't do a build.

RodrigoCard, 2015-12-04:
@electric-monk @FranticRain So, I have some time this week. Do you need some kext sign work? :)
I also want to make a new release with #146 fixes.

larianstudios, 2015-12-04:
Yep, yep :) Release is very soon and link to driver will be right in the game

FranticRain, 2016-01-16:
Resolved in the latest release.

closed 2016-01-16 by FranticRain

pbhogan, 2016-01-17:
Great—thanks for that! Although, now I notice that the mappings are different. I was expecting the mappings to be the same as before (isn't that the case for all controllers supported by this driver?) and just the PID/VID to be correct for identification purposes.

FranticRain, 2016-01-17:
The button numbers are identical, but the packet order is different. If the program you are using subverts the report descriptor, then the buttons show up different. We could re-order the buttons internally, but that means I just did a bunch of work I shouldn't have. I'll prototype you a build and then you can tell me if the buttons show up the same.

FranticRain, 2016-01-18:
Okay, give this a shot and let me know if it works out like you think it should. https://drive.google.com/file/d/0B5PLCKoMd4zGdmJXblYyd1BIN28/view?usp=sharing

pbhogan, 2016-01-19:
Yup! That works as expected.

I really appreciate you looking into all this. I'm the author of [InControl](http://u3d.as/6mr) (a cross-platform controller plugin for Unity) and deal with a lot of developers trying to get controllers working properly in their games — hence my interest in getting this all right. :)

FranticRain, 2016-01-19:
No, that's great. Its nice to have some guidelines on what to do. I've just been freewheeling it for a long time doing what I want, and it just ends up as a hodge podge. Since you might know, whats the best way to get an interface out there for trigger rumble on Xbox One controllers? It looks like there is only one more rumble axis left, per OS X's force feedback spec. Do you have any ideas?

pbhogan, 2016-01-19:
Oh, that's interesting... and unfortunate. 

I haven't dealt with the force feedback spec or frameworks (yet). I'm working on a native module for my plugin now that supports haptics, but currently what I'm doing is using IOKit with HID and use IOHIDDeviceSetReport to do it specifically for devices supported by this driver (which I point people to on Mac) and for the Dual Shock 4.

I suppose if given no other choice you could at least expose the right trigger which I'd think more games use than the left... it's still far from ideal, of course.

That said, if you add trigger haptics and I can trigger (heh-heh) it with a set report, let me know. :)