360Controller/360Controller #176: XBox One controller recognized but test and binding not working.

StickerZombie, 2015-12-01:
Hi, I've successfuly installed Beta3 on my MacBook Pro with Yosemite 10.10.2
My Xbox One controller is recognized by the system profiler and by the driver prefs panel as you can see on the screenshots:
[image]
[image]
But I can't test or re-bind inputs.
The system is somehow detecting the controller as I can use it from Unity (with some weird input names) but the panel is not responding to any inputs.

Help! :_(

FranticRain, 2015-12-01:
Okay first, make sure that you have fully uninstalled any other driver from your system. For example, if you previously had Xone-OSX installed, make sure to run the `uninstall.command` to fully remove it. If you have made sure to fully uninstall all other drivers, and the light on the controller still turns on, then it looks like the driver thinks it is connected, but the controller isn't properly communicating with the driver.

StickerZombie, 2015-12-01:
I did have Xone-OSX previously installed. I don't know how to run the uninstall.command you talk about. Any hints? Thanks!

FranticRain, 2015-12-01:
Download the latest version of the Xone-OSX driver. In the downloaded .dmg file, you will find the `uninstall.command`. Simply double click it and enter your password. It will remove the driver from your system. You may have to reboot before the controller will work properly again.

StickerZombie, 2015-12-01:
OK, worked perfectly, now the test and binding utils are back. The problem is that now Unity has all the inputs messed up :(
Any help with that?
Thanks for all the patience :)

FranticRain, 2015-12-01:
This driver's issues with Unity are fairly well documented in #129. You might also have the same problem as #132. Since I resolved the controller not working part of this, I'm going to close this issue. If you have any more questions about the Unity button mappings, you should probably continue the conversation on #129.

closed 2015-12-01 by FranticRain