360Controller/360Controller #180: Driver doesn't seem to work with 10.11.2b5 or 10.11.2 (final)

jeshuamj2, 2015-12-12:
Driver was working fine with 10.11.2 up to beta 4. After updating to beta 5 and then eventually the final release, the driver no longer recognizes the controller. Is this a known compatibility issue?

Console screenshot immediately after connecting controller.

[image]

jeshuamj2, 2015-12-12:
Update: Oddly enough, it seems to work fine with FranticRain's deprecated driver (Xone-OSX).

slime73, 2015-12-12:
Do you have both installed at the same time? Apparently there may be issues unless you completely uninstall Xone-OSX via `uninstall.command`.

jeshuamj2, 2015-12-12:
No, I was having trouble with just the current one, so I uninstalled it and tried the old one to see if it worked. I'd already had problems in the past and made sure to have the old one uninstalled when I tested this one.

FranticRain, 2015-12-12:
I have 10.11.2 installed, and wired Xbox One controllers work perfectly fine. Do you have something else installed? The console printouts don't look like they are from 360Controller, and I don't recognize them from Xone-OSX either.

chadnielsen, 2015-12-12:
No issues so far with Version 0.15 (beta 3) unofficial and 10.11.2 final with Xbox 360 wireless controller. Early versions of the OS X 10.11.2 beta did give me fits until I went back to 10.11.1.

closed 2015-12-22 by jeshuamj2

lbonfante, 2015-12-30:
Can someone confirm if the OSX 10.11.2 Final version (15C50) works fine and with out problem with the "Drivers360Controller Version 0.15 (beta 3)", i will buy the xbox360 controller, please where can i download the .dmg installer?

FranticRain, 2015-12-30:
@lbonfante This driver works with OS X 10.11.2 just fine. You can download the dmg from the releases page of the repository: https://github.com/360Controller/360Controller/releases

lbonfante, 2015-12-31:
Responding my self, yes works excelent and very well with OSX 10.11.2 and "Drivers360Controller Version 0.15 (beta 3)" download from https://github.com/360Controller/360Controller/releases