360Controller/360Controller #181: Options not saving

jnicol, 2015-12-14:
I am having a problem where any changes I make in the XBox 360 Controllers System Preference pane are lost when I close and reopen the preference pane. e.g If I create a deadzone it disappears when I close the preference pane.

I assume this is not the expected behavior? Is there a "save" button I am overlooking?

- XBox 360 wired controller
- Yosemite
- Tried 0.15 beta 1 and 0.15 beta 3, with the same result
- Tried removing and plugging controller back in
- Tried restarting

jnicol, 2015-12-14:
I just checked the OS X console for errors and here are examples of what I see:

15/12/2015 7:13:25.220 am System Preferences[1501]: Failed to connect (xoneRumbleOptions) outlet from (Pref360ControlPref) to (NSPopUpButton): missing setter or instance variable

15/12/2015 7:13:25.219 am System Preferences[1501]: Failed to connect (linearDeadzoneLeft) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable

I also see two messages below that about may be relevant:

15/12/2015 7:13:25.641 am System Preferences[1501]: No settings found

15/12/2015 7:13:25.727 am System Preferences[1501]: Result of kextstat = 0

RodrigoCard, 2015-12-16:
this one seems related to #67

lawnmowerlatte, 2015-12-17:
I have the same results with the Xbox One controller (via USB). Below are the logs from the issue, I'm not sure if the last three `kernel` messages are related so I've included them for completeness.

    12/16/15 1:42:54.924 PM System Preferences[30738]: Failed to connect (bindingInputColumn) outlet from (Pref360ControlPref) to (NSTableColumn): missing setter or instance variable
    12/16/15 1:42:54.925 PM System Preferences[30738]: Failed to connect (linearDeadzoneLeft) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
    12/16/15 1:42:54.925 PM System Preferences[30738]: Failed to connect (linearDeadzoneRight) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
    12/16/15 1:42:54.925 PM System Preferences[30738]: Failed to connect (xoneRumbleOptions) outlet from (Pref360ControlPref) to (NSPopUpButton): missing setter or instance variable
    12/16/15 1:42:54.930 PM System Preferences[30738]: Unable to simultaneously satisfy constraints:
    (
        "<NSLayoutConstraint:0x6080002855a0 H:[MyWhole360Controller:0x6080001d4eb0(465)]>",
        "<NSLayoutConstraint:0x608000285820 H:|-(NSSpace(20))-[MyWhole360Controller:0x6080001d4eb0]   (Names: '|':NSView:0x6080001307c0 )>",
        "<NSAutoresizingMaskLayoutConstraint:0x60000028b270 h=-&- v=-&- H:|-(0)-[NSView:0x6080001307c0]   (Names: '|':NSTabView:0x7fc552f48440 )>",
        "<NSAutoresizingMaskLayoutConstraint:0x60000028b2c0 h=-&- v=-&- H:[NSView:0x6080001307c0]-(0)-|   (Names: '|':NSTabView:0x7fc552f48440 )>",
        "<NSLayoutConstraint:0x6080002817c0 H:[NSTabView:0x7fc552f48440]-(0)-|   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x608000281810 H:|-(0)-[NSTabView:0x7fc552f48440]   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x608000285870 H:[MyWhole360Controller:0x6080001d4eb0]-(NSSpace(8))-[NSBox:0x7fc552e1e470'Left Stick']>",
        "<NSLayoutConstraint:0x6080002838e0 H:[NSBox:0x7fc552e1e470'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x6080001307c0 )>",
        "<NSLayoutConstraint:0x6080002832f0 H:[NSBox:0x7fc552e1e470'Left Stick'(145)]>",
        "<NSLayoutConstraint:0x6080002815e0 H:[NSBox:0x7fc552f48d00'Box']-(0)-|   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x6080002816d0 H:|-(0)-[NSBox:0x7fc552f48d00'Box']   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x608000280ff0 H:[NSBox:0x7fc552f48d00'Box'(668)]>"
    )
    
    Will attempt to recover by breaking constraint 
    <NSLayoutConstraint:0x6080002855a0 H:[MyWhole360Controller:0x6080001d4eb0(465)]>
    
    Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
    12/16/15 1:42:54.959 PM System Preferences[30738]: Unable to simultaneously satisfy constraints:
    (
        "<NSLayoutConstraint:0x6080002855a0 H:[MyWhole360Controller:0x6080001d4eb0(465)]>",
        "<NSLayoutConstraint:0x608000285820 H:|-(NSSpace(20))-[MyWhole360Controller:0x6080001d4eb0]   (Names: '|':NSView:0x6080001307c0 )>",
        "<NSAutoresizingMaskLayoutConstraint:0x60000028b270 h=-&- v=-&- H:|-(0)-[NSView:0x6080001307c0]   (Names: '|':NSTabView:0x7fc552f48440 )>",
        "<NSAutoresizingMaskLayoutConstraint:0x60000028b2c0 h=-&- v=-&- H:[NSView:0x6080001307c0]-(0)-|   (Names: '|':NSTabView:0x7fc552f48440 )>",
        "<NSLayoutConstraint:0x6080002817c0 H:[NSTabView:0x7fc552f48440]-(0)-|   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x608000281810 H:|-(0)-[NSTabView:0x7fc552f48440]   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x608000285870 H:[MyWhole360Controller:0x6080001d4eb0]-(NSSpace(8))-[NSBox:0x7fc552e1e470'Left Stick']>",
        "<NSLayoutConstraint:0x6080002838e0 H:[NSBox:0x7fc552e1e470'Left Stick']-(NSSpace(20))-|   (Names: '|':NSView:0x6080001307c0 )>",
        "<NSLayoutConstraint:0x6080002832f0 H:[NSBox:0x7fc552e1e470'Left Stick'(145)]>",
        "<NSLayoutConstraint:0x6080002815e0 H:[NSBox:0x7fc552f48d00'Box']-(0)-|   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x6080002816d0 H:|-(0)-[NSBox:0x7fc552f48d00'Box']   (Names: '|':NSView:0x608000130a40 )>",
        "<NSLayoutConstraint:0x608000280ff0 H:[NSBox:0x7fc552f48d00'Box'(668)]>"
    )
    
    Will attempt to recover by breaking constraint 
    <NSLayoutConstraint:0x6080002855a0 H:[MyWhole360Controller:0x6080001d4eb0(465)]>
    
    Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
    12/16/15 1:42:55.224 PM System Preferences[30738]: Unable to simultaneously satisfy constraints:
    (
        "<NSLayoutConstraint:0x6080002855a0 H:[MyWhole360Controller:0x6080001d4eb0(465)]>",
        "<NSLayoutConstraint:0x608000285820 H:|-(NSSpace(20))-[MyWhole360Controller:0x6080001d4eb0]   (Names: '|':NSView:0x6080001307c0 )>",
        "<NSAutoresizingMaskLayoutConstraint:0x60000028b270 h=-&- v=-&- H:|-(0)-[NSView:0x6080001307c0]   (Names: '|':NSTabView:0x7fc552f48440 )>",
        "<NSAutoresizingMaskLayoutConstraint:0x60000028b2c0 h=-&- v=-&- H:[NSView:0x6080001307c0]-(0)-|   (Names: '|':NSTabView:0x7fc552f48440 )>",
[... cut ...]
    Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
    12/16/15 1:42:55.237 PM System Preferences[30738]: No settings found
    12/16/15 1:42:55.400 PM System Preferences[30738]: Result of kextstat = 0
    12/16/15 1:42:59.000 PM kernel[0]: No interval found for . Using 8000000
    12/16/15 1:42:59.000 PM kernel[0]: No interval found for . Using 8000000
    12/16/15 1:42:59.000 PM kernel[0]: No interval found for . Using 8000000

pzerou, 2016-05-23:
Remains to be affecting v0.16 release. Should this be revised for a Milestone aligned w/ 0.16+ version? Settings on deadzone is specifically the item I cannot get saved : (

- 360Controller v0.16
- OS X 10.9.5
- Xbox 360 Wireless Controller

sidenote: Awesome App!

FranticRain, 2016-05-23:
@pzerou This is currently being addressed in #280.

FranticRain, 2018-06-01:
Settings bugs should be fixed in the latest release.

closed 2018-06-01 by FranticRain