360Controller/360Controller #183: Steelseries Nimbus

jeshuamj2, 2015-12-22:
This may be a long shot, but would it be possible to emulate the Steelseries Nimbus controller as an Xbox 360 controller so its usable on Mac? At the current state the d-pad isn't recognized by games.

slime73, 2015-12-22:
The SDL2 library, which a lot of games use, didn't recognize the d-pad of that controller until recently. Now that's fixed though (although there hasn't been an official release with the fix in it yet), so once it's out you could replace the SDL2.framework of games with the new one.

monchote, 2016-01-07:
I'd love to see this too. In case it helps:

```
Vendor ID:	0x0111
Product ID:	0x1420
```

FranticRain, 2016-01-07:
This is not a Xbox 360 controller. It won't work with this driver. Android and iOS controllers are an entirely different beast.

closed 2016-01-07 by FranticRain