360Controller/360Controller #185: Xbox One chatpad not supported

worldwise001, 2015-12-26:
Hello! I have an Xbox One controller + chatpad which I'm trying with this for the first time. The controller appears to be recognized fine, but the chatpad does not work. A quick perusal through some issues seem to hint at the chatpad normally being recognized for the Xbox 360 controller (and I haven't got my 360 controller+chatpad to test). I assume no one has just gotten around to adding support for this?

Some datas:
I am on OS X El Capitan 10.11.2 with what I believe to be one of the newer generation Xbox One controllers. The version of 360controller I have is Version 0.15 (beta 3) unofficial.

According to dmesg, after plugging in the controller I get the following:
Unknown device release 203
start - unable to find chatpad interface

Let me know if there's anything I else I can help with, or any other information I can provide.

FranticRain, 2015-12-26:
Related to #3

1NEGROUP, 2017-07-02:
Request Support For Chatpad.
I am Using the Xbox Elite Controller Which Works Just Fine, But the Microsoft Chatpad Does not.
The Version Of 360Controller is 0.16
Let Me Know if There is Anything I can Do to Help.
You can Contact Me @ https://twitter.com/1NEGROUP
or @ https://www.minds.com/1negroup

Hotmud9, 2018-11-19:
I'm having a similar issue. I'm using 360Controller Version 0.16.5, and my Xbox One controller works just fine, but my chatpad isn't working. My Mac OS Version is 10.12.6.

FranticRain, 2018-11-19:
Xbox One chatpads have never been supported for this driver. Support is not planned at this time.

Hotmud9, 2018-11-21:
F*ck