360Controller/360Controller #186: Controller Not Recognized On Some Games

ashleyjihye, 2015-12-27:
For some reason, the controller isn't being recognized on specific games (even though those games definitely support XBox Controllers). It works fine on Tomb Raider and Bioshock, but I've tried using it on Psychonauts and Grim Fandango, and I can't even get the game to recognize the device.

RodrigoCard, 2015-12-27:
which version are you using?

ashleyjihye, 2015-12-27:
I tried both the 360Controller 0.15 (beta 3) and 360Controller 0.14 releases.

FranticRain, 2016-01-19:
Psychonauts only has partial controller support. Its a bit of a mess on OS X, but it does sort-of work. Grim Fandango has full controller support and should work just fine.

closed 2016-01-19 by FranticRain

ashleyjihye, 2016-01-19:
Do you have any tips on getting it to work with Psychonauts?

FranticRain, 2016-01-19:
Plug in the controller and then use it. It works by default. You can't remap the left stick, but you should be able to remap the buttons from within the start menu of the game. You'll probably need to use a mouse and keyboard to get to that menu though.

ashleyjihye, 2016-01-19:
It doesn't work when I just plug it in. None of the buttons/sticks work at all; only the mouse does.

FranticRain, 2016-01-19:
If you are starting on the brain menu and the game tells you to press space, then: Down on the D-Pad should get you past that prompt. Then, once Raz appears on the brain, you can move around with the left stick. That's just a proof on concept. From there, press escape on the keyboard and access the control bindings menu.