360Controller/360Controller #189: Strange Issue with Broforce on Steam

TillMatthis, 2015-12-30:
Hi,
I'm running El Capitan with Version 0.15 Beta 3 and also tried 0.14 before.
I can map all the Buttons correctly in the setup and the controller works fine in the menu,
but in game the down D-PAD Button triggers another button (chat / enter) and makes the game unplayable. Everything else works fine. Just the down D-PAD doesn't work.
Game description says "full controller support".

Any ideas? Else just see this as a heads up. Thanks for the great work!

Deph0, 2015-12-30:
sound like more of an game problem then driver issue, try contacting the developer about it?

TillMatthis, 2015-12-30:
It works with a generic plug and play usb controller, that has a 360 layout, but doesn't  need a driver. 
This bug only occurs with the 360 controller, but I'll see what I can find out. Thanks.

TyanColte, 2015-12-31:
@TillMatthis this happened to me also, i just remapped the buttons in game to what they're supposed to be and it's working fine now.

FranticRain, 2016-01-19:
If you can remap the buttons in game, this isn't an issue with the driver.

closed 2016-01-19 by FranticRain