360Controller/360Controller #192: Controller Not Recognized

mr-mcox, 2016-01-02:
I have a controller that is not recognized when plugged in (*System Log: No devices found? :( device count==0, i==0*), very similar to [#168](https://github.com/360Controller/360Controller/issues/168#issue-115700352), but with a different vendor ID.

Reading through that issue leads me to think my controller is another bizarre knockoff as the back of my controller looks nearly identical to the [picture posted](https://github.com/360Controller/360Controller/issues/168#issuecomment-155215307). While the product ID is the same, the vendor ID is different. My purpose in posting is to alert maintainers so that this controller can potentially be included when a new release is made based off #151 .

For what its worth, I'm running 10.11.2 and uninstalled and reinstalled Beta 3, but to no avail.

Is there anything else that I can do to assist with the resolution of this issue?

From system information:
Product ID:	0x02a0
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	06962810
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft Corporation.
  Location ID:	0x14600000 / 14
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

FranticRain, 2016-01-02:
Looks like another counterfeit. 0x0e6f is a common vendor ID for these types of things. I could add it and do a build for you real quick, but I honestly don't know what to call it in the info.plist. When you bought it, did it come in packaging similar to a normal Xbox 360 controller? Does it have any marking for any other company on it?

mr-mcox, 2016-01-03:
@FranticRain When I purchased it, I was under the impression that I purchased a genuine Xbox 360 controller, and nothing on the packaging or the [Amazon page](http://www.amazon.com/gp/product/B004QRKWLA) indicated otherwise. I don't see any distinguishing markings. Perhaps it can be called something like "Xbox 360 controller - counterfeit 1", so that when others are found they can be distinguished by that index.

FranticRain, 2016-01-03:
Okay, this should get you fixed up. Let me know if it works so we can close this issue.
https://drive.google.com/file/d/0B5PLCKoMd4zGY21LajVZcjQ1V0U/view?usp=sharing

mr-mcox, 2016-01-03:
That works! Thank you so much!
Matthew

commit f667c1f6 (2016-01-14): Resolves #192
closed 2016-01-03 by FranticRain