360Controller/360Controller #193: Add "Twin USB Joystick" as supported device?

ndroftheline, 2016-01-02:
Hi, 

I would like to request the addition of a generic Twin USB Joystick to the list of devices. Info from System Information app: 

Twin USB Joystick:

  Product ID:	0x0001
  Vendor ID:	0x0810  (Personal Communication Systems, Inc.)
  Version:	1.06
  Speed:	Up to 1.5 Mb/sec
  Location ID:	0x14100000 / 13
  Current Available (mA):	500
  Current Required (mA):	500

Please let me know if I can provide any additional input. 

Thank you very much!

~Micah

ndroftheline, 2016-01-02:
I wanted to add some more detail to this request. I'm using a cheap unbranded PS1/2 controller adapter so I can use my old DualShock 1 and 2 controllers with my Mac. This is very limiting of course to older games that don't require or support ranged/analog triggers, but there are a lot of those so it's OK for my needs. Originally when I found 360controller I read this in the description: "Third party controllers should also be supported, but these must be added manually to the driver's list of supported devices." I thought this app might do something similar to x360ce but for mac. My goal is to use my DualShock 2 masquerading as an Xbox 360 controller for playing games on my Windows machine via Steam in-home streaming. 

I'm comfortable at a command line and editing config files. I've also built software before from source on Linux systems (although not on Mac yet). If I can be of help, please don't hesitate to let me know. 

~Micah

closed 2016-01-02 by ndroftheline

kanytu, 2019-04-08:
Why is this closed?

ndroftheline, 2019-04-10:
I don't remember, sorry. I might have done it manually per instructions here, but I'm only guessing, this was so long ago. https://github.com/360Controller/360Controller#adding-third-party-controllers