360Controller/360Controller #194: Mouse cursor keeps moving left on El Capitan (with xbox one controller)

Rune-keeper, 2016-01-03:
The mouse cursor keeps moving left when I choose the option to use gamepad in games. I have to keep moving my mouse to the right to stop this... I know it is not compatible with El Capitan yet, but is there an easy way to fix it?
Other than this it works great for me. I also use a mapping to keyboard software to set my keys.
Thank you!

Rune-keeper, 2016-01-03:
Maybe the dead zone function is not working properly with El Capitan. There are no other issues other than the mouse cursor moving, and in my games the mouse is mapped to the right stick. Maybe the controller is sending all four directions' signal to the game, and the game preferentially accepts the left signal first.
Just my guess, and I have no knowledge about writing programs...

Rune-keeper, 2016-01-03:
And another phenomenon: pressing LB can reverse the direction (to the right).
I'm not desperate about this :p just want to let you know what I find :p

FranticRain, 2016-01-03:
Just out of curiosity, what games exactly have you noticed this with? Does it happen with all games or just some?

Rune-keeper, 2016-01-04:
You reminds of this. It happens with Euro truck simulator 2 and Farming simulator 2015, but it works properly with Euro truck simulator 1.

FranticRain, 2016-01-04:
The controller maps directly to the mouse in those games. The mouse movement is very much on purpose. For example, in the ETS2 menu, when you are choosing jobs and looking at the map, the joystick should move the mouse cursor. In fact, there is a sensitivity setting for it. This is intended behavior for these specific games, not a bug.

closed 2016-01-04 by FranticRain

Rune-keeper, 2016-01-04:
Yeah, I just changed some options in the games, and now it works nicely with my games. Thank you!